wifidog/wifidog-gateway #45: Support for FQDNs in wifidog.conf

MikyKuroneko, 2015-01-23:
Hello,
I need to whitelist apple's domains to manage the behaviour of Mac OS and iOS when connecting to Wifidog. From what I have read across multiple mailing lists, forums and other issues the thing is that it's actually supported to add domain names to FirewallRules on wifidog.conf, but there's a problem in conf.c that prevents this from working.

I tried compiling an ipk for OpenWRT Barrier Breaker but to no avail. Is there any possibility this could be patched? If so, how can I get this running on a Tp-Link AC1200/AC1750 with Barrier Breaker?

Thank you very much!

mhaas, 2015-03-12:
Fixed in #65.

mhaas, 2015-03-12:
Let me know if you still need instructions on building the ipk.

closed 2015-03-12 by mhaas