wifidog/wifidog-gateway #47: wifidog client dies after 112 simultaneous users

Kvncrck, 2015-02-04:
I git cloned https://github.com/wifidog/wifidog-gateway on Feb 3rd. The last entry in the ChangeLog was from 2014-12-19. I compiled and installed the wifidog client on a debian system with 4 gigs of RAM. Memory usage is not high, but the client freezes once it hits 112 users. No new clients are able to join.  I have to /etc/init.d/wifidog restart to reset the connections so that users can join. But I hit the 112 user limit again. 

I have a trusted Mac list which seems to get truncated in the wdctl status list the closer we come to the 112 th user.

Any ideas?

Thanks

Kevin

Kvncrck, 2015-02-04:
I am using  apAuthSplashOnly and apExternalCMSPlugins.

KazukiShimada, 2015-02-08:
Hi, Kevin.

I'm interested in your problem, too. I took a look at gateway sources.
It seems client_list_append in client_list.c add new client to client list,
https://github.com/wifidog/wifidog-gateway/blob/master/src/client_list.c
 and it is called from http_callback_auth in http.c.
https://github.com/wifidog/wifidog-gateway/blob/master/src/http.c
But I can't find any limitation of adding client.

If my memory is correct, you can get debug message when you start wifidog gateway with option "-d 7". It may show some clue.

Kazuki

Kvncrck, 2015-02-09:
Hi Kazuki

Thanks for your interest in this problem. When I run wifidog -f -d7, new
users are unable to see the authentication page after a while (20 users
today). I can see the download count for existing users increment, however
new users are unable to get to the external splash screen, and "login". If
I kill wifidog -f -d7 and restart wifidog (/etc/init.d/wifidog start) new
users are able to see the splash  page again.  Currently there are 92 new
users since the restart and increasing.

This is the type information I was getting from the debug.

Date: Mon, 09 Feb 2015 14:39:27 GMT
Server: Apache/2.4.10 (Debian)
Set-Cookie: authpuppy=id6ililshj5mcvon8ld3qri0d5; path=/
Content-Lengt(: 7
Connection: close
Content-Type: text/html; charset=utf-8

Auth: 1]
[6][Mon Feb  9 10:39:56 2015][3214](centralserver.c:168) Auth server
returned authentication code 1
[7][Mon Feb  9 10:39:56 2015][3214](firewall.c:258) Locking client list
[7][Mon Feb  9 10:39:56 2015][3214](firewall.c:258) Client list locked
[6][Mon Feb  9 10:39:56 2015][3214](firewall.c:265) Checking client
10.10.1.37 for timeout:  Last updated 1423492792 (4 seconds ago), timeout
delay 139749645877848 seconds, current time 1423492796,
[7][Mon Feb  9 10:39:56 2015][3214](firewall.c:248) Unlocking client list
[7][Mon Feb  9 10:39:56 2015][3214](firewall.c:248) Client list unlocked
[7][Mon Feb  9 10:39:56 2015][3214](centralserver.c:188) Locking config
[7][Mon Feb  9 10:39:56 2015][3214](centralserver.c:188) Config locked
[7][Mon Feb  9 10:39:56 2015][3214](centralserver.c:232) Level 1:
Calculated 1 auth servers in list
[7][Mon Feb  9 10:39:56 2015][3214](centralserver.c:248) Level 1: Resolving
auth server xxx.xxxx.xx]
[7][Mon Feb  9 10:39:56 2015][3214](util.c:140) Locking wd_gethostbyname()
[7][Mon Feb  9 10:39:56 2015][3214](util.c:140) wd_gethostbyname() locked
[7][Mon Feb  9 10:39:56 2015][3214](util.c:155) Unlocking wd_gethostbyname()
[7][Mon Feb  9 10:39:56 2015][3214](util.c:155) wd_gethostbyname() unlocked
[7][Mon Feb  9 10:39:56 2015][3214](centralserver.c:308) Level 1: Resolving
auth server [xxx.xxxx.xx] succeeded = [###.###.###.###]
[7][Mon Feb  9 10:39:56 2015][3214](centralserver.c:333) Level 1:
Connecting to auth server xxx.xxxx.xx:80
[7][Mon Feb  9 10:39:56 2015][3214](centralserver.c:359) Level 1:
Successfully connected to auth server xxx.xxxx.xx:80
[7][Mon Feb  9 10:39:56 2015][3214](centralserver.c:190) Unlocking config
[7][Mon Feb  9 10:39:56 2015][3214](centralserver.c:190) Config unlocked
[7][Mon Feb  9 10:39:56 2015][3214](centralserver.c:197) Connected to auth
server
[7][Mon Feb  9 10:39:56 2015][3214](centralserver.c:114) Sending HTTP
request to auth server: [GET
/auth/?stage=counters&ip=10.10.1.207&mac=60:f8:1d:aa:ec:88&token=713c5754b492eb406c75d6aa8efe920c18af088a&incoming=41984998&outgoing=9112015&gw_id=C
HTTP/1.0
User-Agent: WiFiDog 20130917
Host: xx.xxxx.xx

I didn't see any glaring errors. Looking at the log I was unable to detect
when users started to be prevented from viewing the splash screen.

Another clue? When the 112 user connects and subsequent users are unable to
connect, the entry in the client list is truncated so that it appears
something like this:

Client 112
  IP: 10.10.1.43 MAC: f0:24:75

It is not always truncated at the same position. It almost looks like it is
a byte size limit, rather than number of clients limit. If there were a
limit on the number of users I would expect the last entry to be complete.

I found net.netfilter.nf_conntrack_acct was set to 0 so I set it to 1 and
tried again with the same results.

I tried increasing limits on open files, and tweaking other system limits
e.g number of incoming connections to no avail.

Thanks

Kevin

Kvncrck, 2015-03-10:
Hi,

I believe the command wdctl status is related to the limit on the number of wifi users that we were encountering. If we do not run wdctl status the number of users climbs well above 100 according to the authpuppy web console. Once we run wdctl status with around 100 users, the system seems to freeze up.  The wdctl status output is truncated. Users are unable to view and accept the splash page.

Regards

Kevin

acv, 2015-03-11:
I think the 112 limit is a red herring... The issue there is that `src/util.c:get_status_text()` fills a fix length buffer, it doesn't seem mis-behaved since you actually get the truncated text back. Something else along the way might fail or leave one of the mutex in a locked state.

mhaas, 2015-03-17:
Didn't you fix this, @acv?

acv, 2015-03-17:
Not yet. I got to redo the whole status buffer building one way or another so it doesn't rely on a 16KB buffer, I have some idea but I was waiting on some patches to land.

acv, 2015-04-03:
The combinations of the latest fixes to be merged in should make `wdctl status` almost completely unrelated to the rest of the gateway.