wifidog/wifidog-gateway #48: Cannot satisfy the following dependencies for wifidog - Wrong kernel

proxit24, 2015-02-08:
Hi,

I compiled an openwrt image BB for my TL-WA801ND v2 including the new Wifidog project (overiding the old Makefile with the new one) and when I try to install the wifidog package using opkg install command I got the following error

Installing wifidog (20130917-440445db60b0c3aff528ea703a828b0567293387) to root...
Ignored wifidog md5sum mismatch.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for wifidog:
 *      kernel (= 3.10.49-1-0114c71ed85677c9c1e4911437af4743) *         kernel (= 3.10.49-1-0114c71ed85677c9c1e4911437af4743) *         kernel (= 3.10.49-1-0114c71ed85677c9c1e4911437af4743) *
 * opkg_install_cmd: Cannot install package wifidog.

"uname -a"  command shows i'm using a 3.10 kernel

champtar, 2015-02-08:
hi @proxit24 
wifidog has some kernel module dependencies,
and in openwrt kernel dependencies are strict, they all need to come from the same build.
Just flash a new image.

proxit24, 2015-02-08:
Hi @champtar 

Thank you for your quick response

Do you mean I have to flash my openwrt box with a new image? an then try to install wifidog package?

champtar, 2015-02-08:
yes, flash "your" image, the one you build when you build wifidog

proxit24, 2015-02-08:
Hi again, 

I falshed it using an other image (r42614) and it works now!

Thank you @champtar  for your assistance

here a link to the image  file:///home/ismael/Downloads/openwrt-ar71xx-generic-tl-wa801nd-v2-squashfs-factory%20r42614.bin

hope that will help others

closed 2015-02-08 by proxit24