wifidog/wifidog-gateway #55: Fix duplicate allow rules created by wifidog when a user authenticates

florida63, 2015-02-22:
Reported and corrected bug in DD WRT
http://svn.dd-wrt.com/changeset/20674

To be confirmed.

florida63, 2015-02-23:
@Kvncrck: I saw after issue41 you had wifidog-gateway that works with many user. Could you send me the return of the order.

iptables -t mangle -L

So I can confirm this problem of unnecessary duplication in firewall rule.

In advance thank you

Kvncrck, 2015-02-24:
Here is the return from the command.

iptables -t mangle -L
Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination
WiFiDog_eth1_Trusted  all  --  anywhere             anywhere
WiFiDog_eth1_Outgoing  all  --  anywhere             anywhere
tcpre      all  --  anywhere             anywhere

Chain INPUT (policy ACCEPT)
target     prot opt source               destination
tcin       all  --  anywhere             anywhere

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination
tcfor      all  --  anywhere             anywhere

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination
tcout      all  --  anywhere             anywhere

Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination
WiFiDog_eth1_Incoming  all  --  anywhere             anywhere
tcpost     all  --  anywhere             anywhere

Chain WiFiDog_eth1_Incoming (1 references)
target     prot opt source               destination

Chain WiFiDog_eth1_Outgoing (1 references)
target     prot opt source               destination

Chain WiFiDog_eth1_Trusted (1 references)
target     prot opt source               destination
MARK       all  --  anywhere             anywhere             MAC
28:10:7B:2C:CB:B6 MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
28:10:7B:19:C6:EC MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
28:10:7B:19:C7:82 MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:E4:7A MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:E8:CE MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:E1:B3 MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:E3:49 MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:E3:6A MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:E4:68 MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:E7:58 MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:EC:6C MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:E4:54 MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:EC:54 MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:EA:4D MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:EB:3D MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:E7:72 MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:DA:7B MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:EC:F4 MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:DC:84 MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:EA:2B MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:E6:B4 MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:C1:E3:88 MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:E1:CB MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:EB:DC MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:E1:D9 MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:E3:DB MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:E4:78 MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:E3:19 MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:E4:AC MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:E5:5C MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:E6:28 MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:E6:10 MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:D4:E9:C9 MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
00:80:64:C1:D8:58 MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
01:E8:61:7E:FD:EB MARK set 0x2
MARK       all  --  anywhere             anywhere             MAC
01:50:1A:C5:CE:8D MARK set 0x2

Chain tcfor (1 references)
target     prot opt source               destination

Chain tcin (1 references)
target     prot opt source               destination

Chain tcout (1 references)
target     prot opt source               destination

Chain tcpost (1 references)
target     prot opt source               destination

Chain tcpre (1 references)
target     prot opt source               destination
root@wfgw:~# ----

Thank-you

florida63, 2015-02-24:
@Kvncrck: Thank you,

I deliberately deleted your message because your post match mac adress trusted.

In fact, i want users who are truly authenticated or an account is created.
mask your mac addresses and guards a certain consistency if you have duplicates.

no need to send "Chain WiFiDog_eth1_Trusted".

Kvncrck, 2015-02-25:
We are using apAuthSplashOnlyPlugin for  authentication. Here is the output
with with  over 112 users authenticated.

Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination
WiFiDog_eth1_Trusted  all  --  anywhere             anywhere
WiFiDog_eth1_Outgoing  all  --  anywhere             anywhere
tcpre      all  --  anywhere             anywhere

Chain INPUT (policy ACCEPT)
target     prot opt source               destination
tcin       all  --  anywhere             anywhere

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination
tcfor      all  --  anywhere             anywhere

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination
tcout      all  --  anywhere             anywhere

Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination
WiFiDog_eth1_Incoming  all  --  anywhere             anywhere
tcpost     all  --  anywhere             anywhere

Chain WiFiDog_eth1_Incoming (1 references)
target     prot opt source               destination
ACCEPT     all  --  anywhere             10.10.2.146
ACCEPT     all  --  anywhere             10.10.2.75
ACCEPT     all  --  anywhere             10.10.2.96
ACCEPT     all  --  anywhere             10.10.2.96
ACCEPT     all  --  anywhere             10.10.2.96
ACCEPT     all  --  anywhere             10.10.2.96
ACCEPT     all  --  anywhere             10.10.1.18
ACCEPT     all  --  anywhere             10.10.2.255
ACCEPT     all  --  anywhere             10.10.0.120
ACCEPT     all  --  anywhere             10.10.0.227
ACCEPT     all  --  anywhere             10.10.0.188
ACCEPT     all  --  anywhere             10.10.1.160
ACCEPT     all  --  anywhere             10.10.2.239
ACCEPT     all  --  anywhere             10.10.2.104
ACCEPT     all  --  anywhere             10.10.2.230
ACCEPT     all  --  anywhere             10.10.1.109
ACCEPT     all  --  anywhere             10.10.2.142
ACCEPT     all  --  anywhere             10.10.2.142
ACCEPT     all  --  anywhere             10.10.2.173
ACCEPT     all  --  anywhere             10.10.1.80
ACCEPT     all  --  anywhere             10.10.0.159
ACCEPT     all  --  anywhere             10.10.2.67
ACCEPT     all  --  anywhere             10.10.0.226
ACCEPT     all  --  anywhere             10.10.1.221
ACCEPT     all  --  anywhere             10.10.2.31
ACCEPT     all  --  anywhere             10.10.1.213
ACCEPT     all  --  anywhere             10.10.1.213
ACCEPT     all  --  anywhere             10.10.1.77
ACCEPT     all  --  anywhere             10.10.1.189
ACCEPT     all  --  anywhere             10.10.2.126
ACCEPT     all  --  anywhere             10.10.0.200
ACCEPT     all  --  anywhere             10.10.0.200
ACCEPT     all  --  anywhere             10.10.2.217
ACCEPT     all  --  anywhere             10.10.2.117
ACCEPT     all  --  anywhere             10.10.1.233
ACCEPT     all  --  anywhere             10.10.2.233
ACCEPT     all  --  anywhere             10.10.1.187
ACCEPT     all  --  anywhere             10.10.2.48
ACCEPT     all  --  anywhere             10.10.1.94
ACCEPT     all  --  anywhere             10.10.1.245
ACCEPT     all  --  anywhere             10.10.2.235
ACCEPT     all  --  anywhere             10.10.2.246
[... 502 lines cut ...]
70:56:81:A6:C1:7D MARK set 0x2
MARK       all  --  10.10.2.109          anywhere             MAC
70:56:81:A6:C1:7D MARK set 0x2
MARK       all  --  10.10.2.109          anywhere             MAC
70:56:81:A6:C1:7D MARK set 0x2
MARK       all  --  10.10.2.109          anywhere             MAC
70:56:81:A6:C1:7D MARK set 0x2
MARK       all  --  10.10.2.109          anywhere             MAC
70:56:81:A6:C1:7D MARK set 0x2
MARK       all  --  10.10.1.68           anywhere             MAC
F0:DB:E2:F4:10:61 MARK set 0x2
MARK       all  --  10.10.2.109          anywhere             MAC
70:56:81:A6:C1:7D MARK set 0x2
MARK       all  --  10.10.2.11           anywhere             MAC
20:C9:D0:B7:80:45 MARK set 0x2
MARK       all  --  10.10.1.177          anywhere             MAC
00:23:4E:6C:70:29 MARK set 0x2
MARK       all  --  10.10.0.124          anywhere             MAC
CC:3A:61:7F:94:91 MARK set 0x2
MARK       all  --  10.10.1.105          anywhere             MAC
60:03:08:A0:41:C8 MARK set 0x2

Chain WiFiDog_eth1_Trusted (1 references)....

Regards

florida63, 2015-02-26:
Oh, there is many redundancy

Kvncrck, 2015-02-26:
Could they be hitting the login button multiple times?

florida63, 2015-02-28:
In all cases, there is no reason to add several times the same rule.

acv, 2015-03-14:
One possible fix is to first check if the rule already exists before creating it.

At one point the code always removed access before granting it in the re-auth code path. This was added explicitly in 2004 and removed a few years later probably because it made no sense without comments.

The problem with that approach is a race condition where the users' access is removed and recreated a few (hundreds in some case) milliseconds later.

acv, 2015-03-22:
@Kvncrck Do you frequently use `wdctl restart`?

florida63, 2015-03-22:
@acv: I have done this test

on the auth page I log (1 time) and then sign out, I reconnected and sign out, etc, etc.

the command "iptables -L -t mangle returns me
Chain WiFiDog_br0_Outgoing (1 references)
target     prot opt source               destination         
MARK       all  --  10.63.57.13          anywhere            MAC 00:E0:18:AC:2E:80 MARK set 0x2 
MARK       all  --  10.63.57.13          anywhere            MAC 00:E0:18:AC:2E:80 MARK set 0x2 
MARK       all  --  10.63.57.13          anywhere            MAC 00:E0:18:AC:2E:80 MARK set 0x2

wdctl restart and iptables -L -t mangle
Chain WiFiDog_br0_Outgoing (1 references)
target     prot opt source               destination         
MARK       all  --  10.63.57.13          anywhere            MAC 00:E0:18:AC:2E:80 MARK set 0x2 

wdctl makes the household.

acv, 2015-03-22:
OK, I think I have an idea.

Kvncrck, 2015-03-23:
Wdctl restart did not seem to fix the problems. I would stop wifidog, then start it. Users would have to revisit the Terms of Service page. I would do this often until I discovered that wdctl status seemed to be the cause. Perhaps executing the problematic "wdctl status" right after trying a wdctl restart led me to conclude that wdctl restart was not working.

>> @Kvncrck Do you frequently use wdctl restart?

Kvncrck, 2015-03-23:
I am not sure if this is related to the problem... The wifidog process dies after a period of time, typically 2 - 3 times a day.  I did find OOM errors in the logs. The wifidog gateway  is running Debian and has 8 GB memory. I monitored the processes on the wifidog gateway using ps and top periodically for  a few days. Most of the time the wifidog process consumed very little memory. On one occasion I caught it consuming all available memory. Stopping the wifidog and starting it resolved the problem.

acv, 2015-03-23:
Yeah, we've been working on memory leaks issues, the newest build should be better for that.

acv, 2015-03-23:
It should also avoid the truncation of the wdctl status to ~112 clients due to fixed 16KB buffer.

Kvncrck, 2015-03-23:
Thanks. Is the newest build available now?

acv, 2015-03-23:
You have to compile from the git source at the moment. On debian, you also have to manually install cyassl (also from github sources, with at least `--enable-ecc`) if you want to use HTTPS between gateway and auth server.

florida63, 2015-03-24:
@acv: Always the same conclusion with the last release (by the same test as above).

Kvncrck, 2015-03-24:
I compiled from the git source and manually installed ecyassl  with --enable-ecc. 
The new gateway was speaking to auth server. About 20  users had consented to the TOS and were logged in. When I checked on the system later, I found that the wifidog process had died. I found the following new error in the logs:

wfgw kernel: [38113.926752] traps: wifidog[9645] general protection ip:7fe5ac715a3f sp:7fe5a8c85b80 error:0 in libc-2.18.so[7fe5ac69c000+1a0000]

I was able to restart wifidog, but it looks like I should go back to the previous version.

Kvncrck, 2015-03-25:
I kept running the latest build of wifidog after the general protection faults. I was suspicious of the "wdctl status" command, so I stopped running it. I haven't  seen any more general protection faults since I stopped using wdctl status. I did find that the wifidog process  on occasion would start consuming all available memory again. It crashed about 3 or 4 times yesterday.  I think the OOMs are occurring with the same frequency as the old build. 

I should point out that I have a number of other sites with 50+ simultaneous users where wifidog is rock solid. It is only at the site with a very high user volume (200 - 300 users) that we have problems.

acv, 2015-03-25:
I'd kill for core files or at least backtraces from those crashes.

core files would be problematic due to user data... But a backtrace would be anonymous and absolutely fantastic!

acv, 2015-04-03:
@Kvncrck We've done some work on the authentication code and the duplicate rule issue is probably at least reduced significantly (new code removes the old MARK after changing it.)