wifidog/wifidog-gateway #56: Fix an issue where wifidog will never pass a new token upstream unless a

florida63, 2015-02-23:
Reported and corrected bug in DD WRT
http://svn.dd-wrt.com/changeset/20675

to be confirmed

acv, 2015-03-12:
Something's fishy with that fix.

It seems to fix a symptom (auth for a client already in the client_list does not use new token) without actually putting the new token in the client entry.

acv, 2015-03-12:
Right fix would be to Lock the client list, find client by IP/Mac and update client->token (freeing previous value if not null) just before the call to authenticate.

Ideally by implementing the logic as a `client_list_update_token()` call.

Then auth.c will work regardless.

acv, 2015-03-15:
I suspect the current code is already fixed by 0a7bb07f.

acv, 2015-03-15:
Confirmed.

closed 2015-03-15 by acv

florida63, 2015-03-15:
That explains why I had no problem.

"Good eye"