wifidog/wifidog-gateway #59: use branch and tag, don't change on master

sinkcup, 2015-03-05:
git use branch for write code, use tag for release a version, should not directly change the master.
docs is here:
http://rogerdudler.github.io/git-guide/index.html

suggestion: use branch and tag

benoitg, 2015-03-05:
When we imported SVN we didn't change the model.  The contribution rate caused no incentive to change the flow untill recently, as most contributions were bug fixes.  What you describe in one possible flow.  Another is git flow.

But I think we should just go with develop branch to receive contributions, and master for releases.  Most contributions arrive in the form of pull request, so feature branches are a bit academic anyway.

sinkcup, 2015-03-06:
e...
If you join other open source projects on github, maybe you will find the flow that most developers like.
suggestion: at least use tag for release...

sinkcup, 2015-03-11:
I am doing it ^_^

florida63, 2015-03-11:
@sinkcup: Welcome to the team.

Caution: When you tag with name 1.0.0, 1.1.0; it brings a confusion with the old version of wifidog. It was better to have tag starting with 1.6.0.

example: V1.6.0_alpha1.---> v1.6.0_alpha2 ---> v1.6.0_beta1 ---> v1.6.0_rc1---> v1.6.0 
when do you think?

sinkcup, 2015-03-11:
Oh, sorry, I don't known there was an old version use 1.1.5. 
You are right, we use 3 version numbers. When this project is bigger, maybe we need alpha->beta->rc. 
example: 1.1.5 -> 1.1.6 -> 1.2.0 -> 1.2.1
docs: http://semver.org/

I will change 1.0.0 to 1.1.6, and 1.1.0 to 1.2.0.
Because there is new feature, we should add the second number from 1.1.x to 1.2.0.
If there are bugs, we will use 1.2.1, 1.2.2, 1.2.3.
Next feature will be 1.3.0

florida63, 2015-03-11:
If you prefers Spent 1.2.0 tag, then renames the 1.1.6 tag in 1.2.0_alpha_1. starting point for the construction of the future 1.2.0. In this version of the future we can predict integrate novelty MHAAS or even started working on IPv6.

There were already a lot of changes since the last real realease. Merits of user feedback before a real version of what makes the interest of alpha and beta tags.

1.2.0 will be ok when we'll be able to follow your plan.

In addition it will be sure to tag each adapted to best https://github.com/wifidog/wifidog-gateway/blob/master/configure.in

sinkcup, 2015-03-11:
oh, I known now. I will change the version number in `configure.in` when I add a tag.
use alpha and beta tags is an good idea, but it's a big change vs now "we use master for release".
I have a suggestion: 
first we use only three numbers tag for release, if there are bugs, we release new version, such as 1.2.0 -> 1.2.1 -> 1.2.2; 
after some days, team members are skilled of this git flow, then we try to use 2.0.0-alpha.1 -> 2.0.0-alpha.2 -> 2.0.0-beta.1 -> 2.0.0-rc.1 -> 2.0.0
what do you think about it?
@benoitg @champtar @mhaas

florida63, 2015-03-11:
Pending the opinion of all, you can possibly temporarily remove 2 tags.

We must also take into account that between version 1.1.5 and now there were two other versions using the date only. In my view have more snapshots done in a hurry.

It also appears that the 1.5.0 and 2.0.0 versions was to host a new protocol for wifidog. @benoitg  be more knowledgeable than me.

In all cases, we must reflect on it to remain consistent.

benoitg, 2015-03-11:
I think we can go back to x.y.z format.   Once we release one, we can remove the date-based snapshots from sourceforge, which should fix any package version detection issue the distributions may have.  So yes, release a version as soon as possible, alpha or not.

As for what 2.0.0 was meant for, it doesn't matter anymore, there is no one working on a json based protocol anymore.  Use major and minor versions for whatever you see fit.

acv, 2015-03-11:
Can confirm that I gave up on JSON-ifying and fixing the protocol on my own.

mhaas, 2015-03-11:
Will the distro (openwrt) tools able to handle a version change from yyyymmdd to x.y.z? E.g. will ipkg correctly upgrade patches?

mhaas, 2015-03-11:
Sorry, sent too early. Will ipkg correctly handle upgrades?

sinkcup, 2015-03-12:
@florida63 Yes, we need reflect on it to remain consistent. I have deleted those two tags.
@mhaas I am checking it. 
1. I compile the old version "20090925", then upload it to an openwrt opkg channel, then `opkg install wifidog`;
2. I compile b55096e288460cbc24e89e589deb665b6891aa2e which support "FirewallRule domain", change `Makefile` to version 1.2.0, upload it, then run

    rm /var/opkg-lists/*
    opkg update
    opkg list-upgradable
    opkg upgrade wifidog

to check it can upgrade or not.

sinkcup, 2015-03-12:
I have see openwrt packages code at https://dev.openwrt.org/wiki/GetSource
wifidog package is in 12.09, not in 14.07, we need pull it to github.com/openwrt/packages.
So I compile wifidog package in 12.09, the `Makefile` content:

    include $(TOPDIR)/rules.mk
    
    PKG_NAME:=wifidog
    PKG_VERSION:=20090925
    PKG_RELEASE:=1

    PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.gz
    PKG_SOURCE_URL:= @SF/$(PKG_NAME)
    PKG_MD5SUM:=e3ecacba67a91b6ea3c1072ba6c5a0b4

    PKG_FIXUP:=autoreconf
    PKG_INSTALL:=1

    include $(INCLUDE_DIR)/package.mk

Compile result: `wifidog_20090925-1_ar71xx.ipk`

[image]

mhaas, 2015-03-12:
Is there a consensus on the next version number? I would like to add milestones to some tickets.

sinkcup, 2015-03-12:
then I compile 1.2.0, `Makefile` is : 

    include $(TOPDIR)/rules.mk
    
    PKG_NAME:=wifidog
    PKG_VERSION:=1.2.0
    PKG_RELEASE:=1

    PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.gz
    PKG_SOURCE_URL:=git@github.com:wifidog/wifidog-gateway.git
    PKG_SOURCE_SUBDIR:=$(PKG_NAME)-$(PKG_VERSION)
    PKG_SOURCE_VERSION:=b55096e288460cbc24e89e589deb665b6891aa2e
    PKG_SOURCE_PROTO:=git
    PKG_SOURCE_MIRROR:=0

    PKG_FIXUP:=autoreconf
    PKG_INSTALL:=1
    
    include $(INCLUDE_DIR)/package.mk

Compile result: `wifidog_1.2.0-1_ar71xx.ipk`
Then I try to upgrade, nothing happened.
[image]

So, the result is opkg can't upgrade from yyyymmdd to x.y.z.
But maybe is not important, because wifidog isn't in openwrt 14.07 packages until now, there was not an offical method to upgrade. 
we need pull it to github.com/openwrt/packages, we can use x.y.z
@mhaas @benoitg @florida63

mhaas, 2015-03-12:
I am on it, see #72. Note: wifidog is in OpenWrt, in the oldpackages feed.

Debian has this concept of epochs. Can you try setting the version number to 1:x.y.z in your makefile? I believe opkg supports this.

acv, 2015-03-12:
On the other front of this suggestion, I'd favour using a two-branch develop + master combination once things settle a bit. With PRs being merged to develop.

sinkcup, 2015-03-13:
@acv yes, and not only that, we will use many branches for dev, every issue has a branch, when test is ok, merge it into master, then delete the branch, then release a tag.
follow this docs : http://rogerdudler.github.io/git-guide/index.html

sinkcup, 2015-03-13:
@acv every pull request has a branch too. when accept a pull request, github will show the steps to create branch for  test, then merge into master.

sinkcup, 2015-03-13:
@mhaas I have received a github email, but don't display in this page, you say : Is there a consensus on the next version number? I would like to add milestones to some tickets.
as openwrt 12.09 stable always use wifidog 20090925, most people don't known have to update wifidog, so maybe we don't need to care about "how to opkg upgrade wifidog from yyyymmdd to x.y.z".
We can have a new beginning, use x.y.z, and submit a openwrt packege to openwrt 14.09 package feed.
1.1.6 for these years commit.
1.2.0 for support domain whitelist. (done)
1.3.0 for change popular_servers from hard code to config, or other issue.
2.0.0 for UCI format config (use `/etc/config/wifidog`), it don't compatible with `/etc/wifidog.conf`, so need a main verison number change.
which milestone for openwrt 14.07 procd ? I have try a procd demo, it can run on 14.07, but can't run on 12.09, we should create two script for 12.09 and 14.07? someone known about procd ?

florida63, 2015-03-13:
suggestion for opgk in openwrt

Currently the Openwrt package size is openwrt
"wifidog_20130917-440445db60b0c3aff528ea703a828b0567293387_brcm47xx.ipk"

One can easily have a format:  wifidog_20130917_x.y.z_brcm47xx.ipk.
The date is not necessarily useful but it is expected to retained continuity in the update.

This requires changes to the make in the feed and change our configure.in kept only for the xyz

line
WIFIDOG_VERSION = 20130917
again become
WIFIDOG_VERSION = $ WIFIDOG_MAJOR_VERSION. $ WIFIDOG_MINOR_VERSION. $ WIFIDOG_MICRO_VERSION

sinkcup, 2015-03-17:
Yes, I will create tag, keep 20130917 for opkg update.
new version will like wifidog_20130917-x.y.z_ralink.ipk
I will create tag.
and we need milestone.

PR #101 (2015-03-19): start to use Semantic Versioning, will release 1.2.0 tag after this merge
because "domain whitelist" is a new feature, so we need change the "MINOR_VERSION", should be 1.2.0 .
after many discussions, we are agreeing.
tag is important for us, should not be delayed. we can fork openwrt/packages and commit package for it until we have tags. And milestone will be created after tag.
I will release 1.2.0 tag after this merge, and we can close #59
closed 2015-03-19 by sinkcup