wifidog/wifidog-gateway #61: Talk SSL to the auth server

mhaas, 2015-03-05:
My auth server *only* talks HTTPS. Right now, I use stunnel in client mode to tunnel the connection. This pulls in quite a lot of dependencies.

My feature request/RFC: hook up a small SSL library like PolarSSL or the smaller cyassl. This could be  enabled on compile time. Hooking up [cyassl is quite straightforward](http://www.yassl.com/yaSSL/Docs-cyassl-manual-3-getting-started.html).

Would there be interest in such a feature? Any ideas on that?

acv, 2015-03-11:
It should probably be configurable (default: on) via the configure script. It was rejected initially because of space/performance constraint on the lower end WRT54Gs, 2MB flash and 8MB ram plus the very limited CPU was insufficient to support SSL.

Certificate validation rules can be an issue with TLS. Library should also support TLS v1.2, forward secure cipher suites (e.g. ECDHE) and modern AEAD ci[her modes (i.e.: AES-GCM).

acv, 2015-03-12:
Looks like cyassl generally meets the requirements I said.

sinkcup, 2015-03-13:
I suggest to show two choices in openwrt menuconfig, wifidog and wifidog-xxxssl, if choose wifidog-xxxssl, it depends ssl when opkg install, else don't depends.
like this: https://github.com/shadowsocks/openwrt-shadowsocks
[image]

mhaas, 2015-03-13:
> ![2015-03-13 14 38

I already have that. I will upload it soon.

PR #63 (2015-03-15): Add SSL Client support
This pull request adds optional SSL client support when talking to the auth server. This is useful if your auth server only speaks HTTPS or if you just want to use HTTPS in a post-snowden world.

The patch is tested lightly: it works with my server. The code is heavily based on the [http://www.yassl.com/yaSSL/Docs-cyassl-manual-11-ssl-tutorial.html](CyaSSL tutorial). No guarantee regards "security" given :)

Fixes #61
closed 2015-03-15 by acv