wifidog/wifidog-gateway #68: fw_iptables.c compares string pointer with constants

acv, 2015-03-11:
fw_iptables.c includes the following code several times:

		if (table=="nat") {
			free(mode);
			return;
		}

The behaviour of this code is dependent on the C optimizer and its ability to optimize out duplicate string constants. The core aspect is that the comparison is by pointer. If the C compiler optmized all the uses of the `"nat"` constant to the same memory location and that is what is present in the variable `table` then the comparison indeed behaves as you expect.

The following trivial example will demonstrate this:
    
    #include <stdio.h>
    #include <string.h>
    
    void
    main(void)
    {
        volatile char *test = strdup("nat");
    
        if (test == "nat") {
            printf("Oh really!\n");
        } else {
            printf("Pointer don't match!\n");
        }
    }

In the above example, omitting the `strdup()` will cause the compiler to generate code where the equality is satisfied, with the `strdup()` the equality is not satisfied since a new pointer is allocated on the heap.

Use `strncmp()` instead.

mhaas, 2015-03-11:
Good catch. I have already fixed this in my ipset pull request, I believe. Perhaps we should set -Wall by default.

acv, 2015-03-12:
Pull request #69 resolves this issue and adds -Wall.

mhaas, 2015-03-12:
Merged #69. Thanks!

closed 2015-03-12 by mhaas