wifidog/wifidog-gateway #72: Bring wifidog back to OpenWrt

mhaas, 2015-03-12:
Wifidog is in the oldpackages feed. For 14.07, binary packages are still available. Newer OpenWrt releases won't be building these anymore.

I already have an updated Makefile including a wifidog-ssl package. Openwrt also uses procd as init which can do process supervision. I have updated the init script accordingly.

This issue is mainly a reminder for me and as well as a place for discussion, should there be need.

florida63, 2015-03-12:
It will reflect on #22 (choice of feed).

KazukiShimada, 2015-03-12:
very very nice!
I want to try it.

sinkcup, 2015-03-13:
Great!
We need use tag for release and anonymous can compile it. otherwise user can't compile it, because it will auto clone code from github, it need a github accout. 
I have try a procd demo, it can run on 14.07, but can't run on 12.09, we should create two script for 12.09 and 14.07? someone known about procd ?

sinkcup, 2015-03-13:
two methods:
1. we create a repo "wifidog-openwrt-package", then link it to openwrt/packages as an submodule http://git-scm.com/book/en/v2/Git-Tools-Submodules
2. fork openwrt/packages, change it, then pull request.

I think 1 is better, but openwrt project maybe don't accept.

mhaas, 2015-03-13:
I think we can just fork openwrt/packages with the team. We all have access then.

Openwrt 12.09 is probably not that relevant. I don't think the openwrt project provides updated packages so we can focus our efforts on 14.07 and trunk. Perhaps mainly trunk, because as with 12.09 the Openwrt guys probably won't update.

I have a working init script as well.

sinkcup, 2015-03-13:
I find some info:
http://wiki.openwrt.org/about/history

Barrier Breaker 14.07	2014 October	Barrier Breaker	r42625
Attitude Adjustment 12.09	2013 April	Attitude Adjustment	r36088

on  https://dev.openwrt.org/wiki/GetSource , two stable version 12.09 and 14.07 are listed (maybe recommended?).
Because of 14.07 is a new version release at 2014, smart router companies are still using 12.09, maybe they will upgrade to 14.07 in next 2 years. So, We may should support 12.09 and 14.07.
"Smart router companies" means offical supply openwrt rom with high-performance driver (maybe privite, like Nvidia GPU driver on Linux), don't need customers to flash 3rd rom which may don't having good driver. These companies are rare, I know a little (using ralink/MTK CPU).

mhaas, 2015-03-13:
Okay, then let's use the existing packaging scripts (Makefile) for 12.09 and update the Wifidog version there. For 14.07 and onward, we can use the new and improved scripts.

sinkcup, 2015-03-13:
Is there a solution for a package both support 12.09 and 14.07, maybe some "define or auto check version" or choose it in menuconfig like "wifidog-for-12.09-and-older, wifidog-for-14.07-and-newer" ?
If not, we need develop two project. :(

databeille, 2015-03-15:
I have a working Makefile for BB. Dealing with AA or BB could be done inside the Makefile. I'll give an hand on this tomorrow ;)

mhaas, 2015-03-17:
The situation is a bit of a mess right now, with @acv merging #97 and @sinkcup reverting in #101.

Maybe we should start over. I like @databeille's contribution, except for some minor niggles. I propose the following:

* Fix any issues with @databeille's contribution (pkg version, init script creation, proper init script vs wifidog-init (?))
* Merge into wifidog in contrib/ and possibly throw out the old scripts in there
* Make sure everyone is happy with the script
* Fork openwrt/packages into the wifidog project
* Bring over our changes (Note: OpenWrt requires a maintainer line in the Makefile)
* Submit PR to OpenWRT

That we, we can always ship our canonical Makefile and also assume maintainership of the OpenWrt package.

databeille, 2015-03-17:
@mhaas : first point should be met https://github.com/wifidog/wifidog-gateway/pull/102

mhaas, 2015-03-18:
Pushing to 1.3.0 as we actually need a release before we can submit to OpenWrt.

sinkcup, 2015-03-19:
for CC, should fork https://github.com/openwrt/packages, commit a new package to master
We are agree with it. Who have the permission to fork to wifidog team, I don't have, see:
[image]

for BB, @mhaas maybe we can fork https://github.com/openwrt/packages, commit a new package to branch for-14.07,and tell admin to delete oldpackages? I think wifidog was in "oldpackages" is because of it's not useing procd, if we update the code, it should in "packages" not "oldpackages", and "oldpackages" is disabled in openwrt sdk `feeds.conf.default`.
see oldpackages http://git.openwrt.org/?p=packages.git;a=tree;f=net/wifidog;h=28e99e20192d9cca8f1bc632c9157ce26dba2bf4;hb=HEAD

for AA, find the way to update old code at https://dev.openwrt.org/browser/branches/packages_12.09/net/wifidog?order=name

benoitg, 2015-03-19:
Forked to wifidog

champtar, 2015-03-19:
hi all,

@sinkcup, you need to make PR against https://github.com/openwrt/packages to have your packages  in openwrt, updating your fork will do nothing.
No one will update AA packages, it's deprecated.
For BB the policy is bugfix/securityfix only, no new packages (but sometimes ...)
for the "oldpackages" vs "packages", it's not related to procd, new "packages" repo need for each packages a maintainer that take care of updating and testing.

To sum up, you need:
1) to send a pull request for openwrt CC (https://github.com/openwrt/packages)
2) create a new repo like i asked in #107, to put AA an BB makefile (i will write a readme to explain how to use it)
3) drop https://github.com/wifidog/packages

mhaas, 2015-03-19:
@champtar 

Re points 1) and 3): the idea was that the team collaboratively works on the packaging. We then submit a pull request to openwrt/packages.

We will still have a single maintainer and squash our commits into one for the PRs. And fulfill the other requirements.

sinkcup, 2015-03-19:
I write a wiki for how to release tag and write openwrt package. @mhaas @benoitg @acv @florida63
https://github.com/wifidog/wifidog-gateway/wiki

julor, 2015-03-24:
wifidog could not start Automatically.
I have used '/etc/init.d/wifidog enable',and found S65wifidog in '/etc/rc.d/'
could you help me?

mhaas, 2015-04-17:
Here we go:
https://github.com/openwrt/packages/pull/1144

Perhaps @champtar would like to take a look on the OpenWrt side of things.

KazukiShimada, 2015-04-17:
@mhaas
Thanks a lot. It's a nice day today.