wifidog/wifidog-gateway #74: Code Quality: run clang static analyzer

mhaas, 2015-03-12:
Once we have all PRs merged, run a static code analyzer like clang to find some more questionable code. On my Ubuntu box, this is as simple as installing clang and running ./autogen.sh && scan-build make.

I have already found a nice error in https://github.com/mhaas/wifidog-gateway/commit/3b27606f018da7464005a020a1861fe0fadd9391.

I'd like to wait for the PRs to be merged because some errors are in centralserver.c and ping_thread.c which I have changed in my SSL pull request.

sinkcup, 2015-03-13:
and we need choose a popular c coding standard.

champtar, 2015-03-13:
hi @sinkcup @mhaas,
may i suggest using travis, to run some test on each pr, commits, ...
see https://github.com/nodogsplash/nodogsplash/commit/0398da2ff59524fec2cfeb421c4f56976e232634

mhaas, 2015-03-13:
Sounds good. We still need a standard, though. Maybe some benevolent dictator will step up and decide for one.

acv, 2015-03-13:
Be careful there... Either Benoit or I could make a pretty good claim at the BDFL title ;-).

I'm flexible but I would suggest:

spaces, not tabs
4 spaces per level

basically:

    set expandtab
    set shiftwidth=4
    set softtabstop=4
    set tabstop=4

In Vim speak.

For function definition, we should pick a template, something like:

    return type *
    my_func(.....) {

Although I don't care whether the { is on the same line or the next line. I dislike the weird GNU braces being half an indent step from code thing.

mhaas, 2015-03-13:
Sounds good. I have the same settings in my vimrc and have to disable it every time for wifidog.

acv, 2015-03-13:
I suggest that we do this as soon as possible after your SSL patch gets merged.

sinkcup, 2015-03-14:
Yes! 4 spaces is good, I'm using it in my projects for many years.

acv, 2015-03-14:
I suggest then the general style obtained with GNU indent with this command-line:

    indent --linux-tyle --no-tabs --indent-level 4 --line-length 120

The length limit works well for me and the existing code but some folks might prefer shorter.

mhaas, 2015-03-15:
What's the guideline for comments and function documentation?

acv, 2015-03-15:
try to keep the existing oxygen comment up-to-date and use the same format for new code...

    /** @brief  description */   // Before declarations in headers.
    /**<@brief description */   // After members in struct
    /** Description
     * @param x  Description
     * @return description
     */    // Before functions in .c files.

acv, 2015-03-15:
Adding `--procnames-start-lines` to the indent option.

Since I prefer this style:

    static type ***
    crazy_func(void ****)
    {

Easier to grep for too `^crazy_func`

acv, 2015-03-15:
I applied this to my libhttpd 1.4 branch.

mhaas, 2015-03-18:
Moving this back to 1.3.0. I'd envision some kind of script or automated environment (Hudson). But we already got some nice improvements out of this for 1.2.0.

mhaas, 2015-03-26:
I think we are done here. Scan-build comes back clean for me and we have Travis + coverity.

closed 2015-03-26 by mhaas