wifidog/wifidog-gateway #78: move popular_servers from hard code to config

sinkcup, 2015-03-13:
now `src/centralserver.c` has a hard code "popular_servers", it should in `wifidog.conf`, then user can change it very easy.
In some country (e.g. China), users can't access www.google.com, and it had been "DNS cache poisoning", so can't use it for DNS health checing.

acv, 2015-03-22:
Do you have time to implement this? Sounds like a great idea.

sinkcup, 2015-03-22:
I will do it in this week, before next tag release

sinkcup, 2015-04-09:
I have finish it, should I move this to milestone 1.2.1 ?
If not, after some commits, it will be hard to merge.

mhaas, 2015-04-09:
Sounds good, just make a pull request.

sinkcup, 2015-04-09:
CI is great!
this pull request CI have been successful, I should not merge my code, someone review and merge it, will be better.

sinkcup, 2015-04-09:
I forgot to show screenshot
[image]
[image]
[image]

closed 2015-05-08 by sinkcup