wifidog/wifidog-gateway #80: Facebook does not work as Auth provider on iOS 8

mhaas, 2015-03-14:
I am using Facebook to authenticate users. On iOS 8, the connection assistant pops up and shows my Auth sever. On clicking "login with facebook", Facebook does not load.

I assume this has something to do with OCSP. This ticket is mainly intended to document the issue. A fix would likely consist of a documentation update with a workaround.

mhaas, 2015-03-17:
So, this had nothing to do with ios8 at all. I forgot to whitelist fbstatic-a.akamaihd.net in dnsmasq. So it got stuck on loading the CSS. I never noticed during my tests because I had these cached.

For reference, the ipset line in dnsmasq.conf looks like this:

    ipset=/facebook.com/fbcdn.net/fbstatic-a.akamaihd.net/crl3.digicert.com/crl4.digicert.com/ocsp.digicert.com/cacerts.digicert.com/fb

closed 2015-03-17 by mhaas

sinkcup, 2015-03-19:
this will put facebook to whitelist ?

mhaas, 2015-03-19:
@sinkcup Yes, if you put this in dnsmasq.conf AND add a "allow to-ipset fb" rule in your wifidog.conf.

sinkcup, 2015-03-19:
...
It looks like I had do something useless, I change to code to support domain whitelist, because I don't know I can use "to-ipset".

mhaas, 2015-03-19:
No, not at all. This requires support for ipset in the DNS forwarder and additional configuration. Your domain whitelist works out-of-the-box, but doesn't do HTTPS.

dirkam, 2015-06-10:
@mhaas 
Does ipset do HTTPS?
How does it actually work? Does it resolves the domains in ipset to IPs upon wifidog initalization?

mhaas, 2015-06-11:
An ipset defines a set of IPs. This set is populated by dnsmasq every time a DNS lookup is made for facebook.com. As it is concerned with DNS, it does not care about HTTP or HTTPS.

dirkam, 2015-06-14:
Is it possible to use this on OpenWRT? It seems that dnsmasq does not support ipsets directly (fails to start if one is defined).

mhaas, 2015-06-14:
Try installing dnsmasq-full or build a dnsmasq package with ipset support. It's quite easy to do with the buildroot/sdk.

dirkam, 2015-06-17:
Did you manage to get this work on OpenWRT or are you using another distro? Even dnsmasq-full fails: failed to create IPset control socket: Protocol not supported

mhaas, 2015-06-21:
I am using OpenWrt BB 14.09. Did you create the ipset before starting dnsmasq?

dirkam, 2015-06-21:
I've tried to copy your example to dnsmasq.conf and restart dnsmasq. Should I declare it somewhere else, too?

mhaas, 2015-06-21:
See https://github.com/mhaas/fbwlan/#allowing-access-to-facebook

But stop following the directions when it comes to the creation of */etc/init.d/wifidog-fw-extra* - it's not needed anymore.

mhaas, 2015-06-21:
Short answer for those playing along at home: You need to add something like

ipset create fb hash:ip

to your /etc/rc.local/

dirkam, 2015-06-21:
Okay, I see. Thanks for the help, will check this later.

dirkam, 2015-06-21:
I think that there is still something missing.
With dnsmasq-full package there is no ipset command available. I installed ipset package as well but I get this error:
ipset v6.20.1: Cannot open session to kernel.

Any idea how could I resolve this? I'm using BB 14.07 as well.

mhaas, 2015-06-22:
I got a notification that you posted another message, but now it's gone? Anyway, Try installing kmod-ipt-ipset on OpenWrt :)

dirkam, 2015-06-22:
Sorry for that.

It seems that there is still something missing. With dnsmasq-full and kmod-ipt-ipset there is no such command as ipset.
If I add the ipset line to dnsmasq.conf and restart it then dnsmasq itself works but I don't think that it's OK as wifidog does not allow connection to the given hosts.
Tried to add ipset create fb hash:ip to either /etc/rc.local or /etc/firewall.user but no success.
I also tried to install ipset and ipset-dns packages. With this I get the following error:
ipset v6.20.1: Cannot open session to kernel.

What do I do wrong?

Btw how exactly are requests handled with ipset? Does it only allow the connection if there is a DNS request from the client? If so then it might be an issue if the user has a cached address for the given hosts (let's say using mobile data connection) and tries to connect to that with the cached address.

mhaas, 2015-07-05:
Have you tried installing all packages:
* kmod-ipt-ipset
* dnsmasq-full

You don't need ipset-dns.

mhaas, 2015-07-05:
Perhaps you also need to load the ipset kernel module.

amosipov, 2015-07-10:
@mhaas, thanks for the new ipset feature, works really well with https, great job!
To make it work I had to additionally install **ipset**, **kmod-ipt-ipset** and **dnsmasq-full** packages with all their dependencies, plus changed **/etc/firewall.user**, **/etc/dnsmasq.conf** and **/etc/wifidog.conf** files according to your posts. The device is TP-LINK TL-WDR3500 on OpenWRT Barrier Breaker 14.07.

@dirkam, how did you install the packages? On the device using opkg or during the built? kmod-ipt-ipset worked properly only after reboot if I used opkg.

dirkam, 2016-01-11:
I tried this again, but I still can't make it work. Can you please let me know what I do wrong?
**ipset, kmod-ipt-ipset** and **dnsmasq-full** packages are installed.
`ipset=/.../fb` added to **/etc/dsnmasq.conf**
`ipset create fb hash:ip` added to **/etc/firewall.user**
`FirewallRule allow to-ipset fb` added to **/etc/wifidog.conf**
`ipset list` shows IP addresses for the domains.
I can see the rule in iptables for the ipset
`ACCEPT     all  --  anywhere             anywhere             match-set fb dst`
However wifidog still intercepts the connection
```
Mon Jan 11 22:08:43 2016 daemon.info wifidog[1968]: Check host fb.com is in whitelist or not
Mon Jan 11 22:08:43 2016 daemon.info wifidog[1968]: rule mask 0.0.0.0/0
Mon Jan 11 22:08:43 2016 daemon.info wifidog[1968]: host fb.com is not in 0.0.0.0/0, continue
Mon Jan 11 22:08:43 2016 daemon.info wifidog[1968]: rule mask 0.0.0.0/0
Mon Jan 11 22:08:43 2016 daemon.info wifidog[1968]: host fb.com is not in 0.0.0.0/0, continue
Mon Jan 11 22:08:43 2016 daemon.info wifidog[1968]: rule mask 0.0.0.0/0
Mon Jan 11 22:08:43 2016 daemon.info wifidog[1968]: host fb.com is not in 0.0.0.0/0, continue
Mon Jan 11 22:08:43 2016 daemon.info wifidog[1968]: rule mask 0.0.0.0/0
Mon Jan 11 22:08:43 2016 daemon.info wifidog[1968]: host fb.com is not in 0.0.0.0/0, continue
Mon Jan 11 22:08:43 2016 daemon.info wifidog[1968]: rule mask 0.0.0.0/0
Mon Jan 11 22:08:43 2016 daemon.info wifidog[1968]: host fb.com is not in 0.0.0.0/0, continue
Mon Jan 11 22:08:43 2016 daemon.info wifidog[1968]: Captured 172.20.0.226 requesting [http%3A%2F%2Ffb.com%2F] and re-directing them to login page

sinkcup, 2016-01-12:
@dirkam You can try another method:

```
FirewallRuleSet global {
FirewallRule allow to fb.com
FirewallRule allow to facebook.com
FirewallRule allow to fbcdn.net
}
```

mhaas, 2016-01-30:
@dirkam Can you show us your dnsmasq.conf?

@sinkcup That won't work - the DNS will be resolved only once, at startup. This does not take into account the DNS rotation used by facebook. In other words, this will only allow one IP while the hosts resolve to multiple IPs.

kuzjnvn, 2017-02-05:
@amosipov can you email me new wifidog ipk for ar71xx openwrt 14.07

amosipov, 2017-02-06:
@kuzjnvn sure, just give me your email

kuzjnvn, 2017-02-07:
@amosipov my email is baolinhtv@gmail.com, thank you so much