wifidog/wifidog-gateway #84: fw_sync_with_authserver needs rewriting

acv, 2015-03-14:
As it runs right now, it's complicated and ultimately there's a number of sections where it makes giant leap of faith that a `t_client` pointer is still valid while it didn't hold the lock on the client list.

I intend to fix this by:

1. Adding a function to `client.c` to validate that a pointer still exists in the client list and still has the same IP/Mac. This would allow us to avoid the leap of faith "does client still exist?"
2. Refactor the whole thing to run in two pass: (a) get a list of tuple `(pointer, ip, mac)` for all clients; (b) iterate over the created list verifying existence for each.

The reason for point 2 is that right now, we resume iterating over the linked list after not holding the lock. There's no guarantee that our `t_client` still exists, that it's `->next` pointer is still valid and so on.

mhaas, 2015-03-18:
Pushing to 1.3.0, if you don't mind.

acv, 2015-03-29:
Adding the bug label because @mhaas' fuzzing with a load tester has uncovered use-after-free and other race conditions in this function.

mhaas, 2015-03-29:
Let's continue the discussion/my ramblings here. Now that I read your bug and look at the backtrace line numbers properly, I too see the problem. I think I can patch it reasonably quick.

acv, 2015-03-29:
That code is the dirty underbelly of the gateway... It needs full refactoring.

acv, 2015-03-29:
If it has *k* clients, it needs to do *k* network calls which can literally take forever. This is why it drops the mutex so just making the whole thing mutex protected is unacceptable.

mhaas, 2015-03-29:
I'm thinking of adding a get_client_list_array function to client_list.c. That is the easy part, but how to prevent double free.. I am thinking of having that function do the cleanup of the client list. A client is marked as gone by setting mac == token == IP == null and then free'ing the memory in fw_sync_with_authserver.

mhaas, 2015-03-29:
If we had a hashset or a self-growing vector data  structure, we could easily store a set of pointers and delete those. Hum. That would be better.

acv, 2015-03-29:
I was lazier. I duplicate the whole client list. And work on the duplicate. Then I just destroy the entire duplicate.

acv, 2015-03-29:
The annoyance with Java style HashSet is, well, how much work they are in C!

I was thinking of keeping the list structure and adding hashed indices.

Right now the list must be searchable by IP, mac OR token.

mhaas, 2015-03-29:
Duplicating the whole list uses more memory, I guess.

I have a is_client_still_valid method that just looks through the list and compares pointers. Your method has the advantage that client identity is checked. I just check for valid pointers.

acv, 2015-03-29:
The memory usage is not unreasonable... And, crucially, it completely sidesteps the validity of `->next`. In the copy, `->next` validity is an invariant. In the original list, it is never assumed unless the mutex is held.

I actually want to move most of the consumers to a DB-like model, where the consumer requests a subset of the list, goes through it and the consumer can do a moral equivalent of an SQL update.

One way I thought of doing this was to assign a unique, 64-bit, ID to all client. That ID would be an invariant and provide a reliable way of saying "this client". 64-bit because there aren't enough nanoseconds left before the sun goes supernova for the counter to wrap around and collide.

acv, 2015-03-29:
Using a DB-like model would have a lot of benefits:

* Easier to reason about concurrency
* Current code assumes either IP or Mac can't repeat. Reality is very different
* The low-level linked-list is no longer exposed so it could be complemented by other data structures to make searching faster

Cons:

* A lot more heap allocations. Lots more potential for leakage. Most of them however will be of the trivial to find by static analysis so coverity will be useful.
* Some code paths will have hard choices to make when confronted with result sets who's length is not 0 or 1. What do you do when there's already 3 clients with that MAC? Or an IP now has a new MAC?

acv, 2015-03-29:
I'm thinking of changing all the find functions to signatures like:
```C
int
client_sublist_findXXXXXX(t_client **result, ....)
```
Where the return value would be number of items found and the pointer pointer passed as input would get `NULL` or a sub-linked list of duplicates. List to be disposed off using `client_list_destroy()`.

mhaas, 2015-03-29:
Yeah, that sounds sane. Just looked at the code, looks good. Especially once the client_update_from_copy (or whatever) function is in. Nice.

acv, 2015-03-29:
I'm actually going to settle for merely calling `client_list_find_by_client()` to obtain the real entry for a given cloned client and then update it while the mutex is held. It's close enough and much simpler then trying to merge changes.

acv, 2015-04-01:
This is now done. Not to say it won't get revisited but the main races should all be gone now.

closed 2015-04-01 by acv