wifidog/wifidog-gateway #87: Resync libhttpd

mhaas, 2015-03-15:
Upstream libhttpd is 1.4 and we are at 1.3. Maybe that will fix some compiler warnings?

http://www.hughes.com.au/products/libhttpd/

acv, 2015-03-15:
HAHAHAHA.

Sorry but no!

Back in 2004 I did this change: 93c94bccd49f22489929ad3466811fcb6cefdfe8

And that broke compatibility with upstream in a big way.

Upstream had the following problems:

1. All parsing was done with fixed length buffers and no bound checking (buffer overflows everywhere.)
2. Not threaded/threadable at all.

florida63, 2015-03-15:
Just for information: Nodogsplash seems used 1.4.
https://github.com/nodogsplash/nodogsplash/commit/f19ef422bf1207e2921e2a875842191c0edcd860

acv, 2015-03-15:
Ah, so someone went through the trouble of reapplying the wifidog patches to upstream. I'm not sure of the benefit... But we could merge their update in.

acv, 2015-03-15:
In fact, switching to the nodogsplash version of libhttps would be good, they fixed things we missed...

f2b83e2d311305e0022de2cccfa3f1ba07176a3e

acv, 2015-03-15:
This is the diff between upstream 1.3 and upstream 1.4 (excluding `configure`):

https://gist.github.com/acv/22ff3b4a64c8b10dc9

Most of the new code needs refactoring to deal with the thread split. The Ember support is another lib from Hughes to embed a tiny scripting language.

acv, 2015-03-15:
I think the easiest way forward is to pick from the diff and integrate what makes sense from it and then bump the version. Also go on a fishing expedition in the nodogsplash repo and cherry pick more fixes.

acv, 2015-03-16:
So, should we bring in the Text Expansion and Ember support from libhttpd 1.4? Both features seem to be geared to using Ember which is Hughes' embedded web scripting language (which only works with libhttpd and mSQL...)

I haven't in my PR #91

mhaas, 2015-03-16:
No. I have only 4MB flash. :)

acv, 2015-03-16:
Cool, then it's pretty complete.

mhaas, 2015-03-16:
Then let's merge? I haven't tested it, but I assume you did ;)

acv, 2015-03-16:
Somewhat ;-).

acv, 2015-03-16:
I'm not nearly as bad as in the early days where if gcc didn't complain, I commited ;-)

mhaas, 2015-03-17:
Groovy! I would have merged via github, but it can't be merged automatically. :(

acv, 2015-03-17:
Merged master back into it. I hope I didn't accidentally remove any code.

mhaas, 2015-03-18:
Fixed in#91.

closed 2015-03-18 by mhaas