wifidog/wifidog-gateway #90: Opening the firewall if the auth servers are all unreachable

florida63, 2015-03-15:
see (http://dev.wifidog.org/ticket/92).

acv, 2015-03-15:
From source:

> The last 2 attachments are a patch from Alex Nault from Zapbsl, compiled and tested by gbastien.
> 
> Works fine, but when the auth server is down and the auth-is-down ruleset is empty, then it does not redirect to the Uh ho! message page as it should do. If anyone has any idea why it doesn't, let me know. The code is not committed yet.

florida63, 2015-03-15:
I had seen but it's still boring to know of this patch, even if we do not integrate. 
In my view, we must refocus bugs and patch present on the old wiki.

acv, 2015-03-15:
I wasn't shooting down the patch, just adding this as a requirement for a PR.

acv, 2015-03-19:
The pull request is basically complete but could use more testing.

mhaas, 2015-03-19:
Does this require an update to doc/wifidog_firewall_map.dia?

acv, 2015-03-19:
Maybe, I don't have dia so I can't see (I always preferred OmniGraffle, which is what I used for the original flow diagram.)

acv, 2015-03-23:
Fix merged but leaving issue open as reminder to update NEWS file tonight.

acv, 2015-03-24:
NEWS updated, all done here.

closed 2015-03-24 by acv