wifidog/wifidog-gateway #99: execute gw_ip:2060/wifidog/auth?token=uuid client surf the Internet  but execute gw_ip:2060/wifidog/status exist client

eggbucket, 2015-03-17:
execute gw_ip:2060/wifidog/auth?token=uuid client surf the Internet  
but execute gw_ip:2060/wifidog/status exist client

databeille, 2015-03-17:
It looks more like a server problem.

eggbucket, 2015-03-17:
this mac  "f0:de:f1:d8:3f:3d" No Internet access

Come back to redirect to "http://server/login "

It looks more like a iptables 'mark' problem.

openwrt:  kernel 2.6.36    iptables:1.4.20  

 this wifidog  run in "kernel 3.10.36"  normal

---------------------------------------------------------------------------------------
WiFiDog status

Version: 20130917
Uptime: 0d 0h 1m 7s
Has been restarted: no
Internet Connectivity: yes
Auth server reachable: yes
Clients served this session: 1

1 clients connected.

Client 0
  IP: 192.168.1.203 MAC: f0:de:f1:d8:3f:3d
  Token: 111111111111111111
  Downloaded: 707877
  Uploaded: 390064

admin@DFC:~# iptables -t mangle -L -n
Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination         
WiFiDog_br-lan_Trusted  all  --  0.0.0.0/0            0.0.0.0/0           
WiFiDog_br-lan_Outgoing  all  --  0.0.0.0/0            0.0.0.0/0           
fwmark     all  --  0.0.0.0/0            0.0.0.0/0           

Chain INPUT (policy ACCEPT)
target     prot opt source               destination         

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination         
mssfix     all  --  0.0.0.0/0            0.0.0.0/0           

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination         

Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination         
WiFiDog_br-lan_Incoming  all  --  0.0.0.0/0            0.0.0.0/0           

Chain WiFiDog_br-lan_Incoming (1 references)
target     prot opt source               destination         
ACCEPT     all  --  0.0.0.0/0            192.168.1.203       

Chain WiFiDog_br-lan_Outgoing (1 references)
target     prot opt source               destination         
MARK       all  --  192.168.1.203        0.0.0.0/0            MAC F0:DE:F1:D8:3F:3D MARK set 0x2

Chain WiFiDog_br-lan_Trusted (1 references)
target     prot opt source               destination         

Chain fwmark (1 references)
target     prot opt source               destination         

Chain mssfix (1 references)
target     prot opt source               destination         
TCPMSS     tcp  --  0.0.0.0/0            0.0.0.0/0            tcp flags:0x06/0x02 /* wan (mtu_fix) */ TCPMSS clamp to PMTU

databeille, 2015-03-17:
Did you manually change the token before posting your last message ?
<code>
 Token: 111111111111111111
</code>

mhaas, 2015-03-18:
We'll need the output of wifidog -d 7, i.e. with high debug level enabled.

eggbucket, 2015-03-18:
thanks,Problem has been resolved 
causal software conflict

closed 2015-03-18 by mhaas