wifidog/wifidog-gateway #103: autogen.sh eats $1 instead of passing to configure

mhaas, 2015-03-18:
If you run

    ./autogen.sh --enable-cyassl

then *enable--cyassl* gets saved to *$BUILDROOT* and weird stuff ensues.

    if [ "X$1" != "X" ]
    then
        BUILDROOT=`echo "$1" | sed 's/^[^=]*[=]//'`
        [..]
       CC=${BUILDROOT}/build_mipsel/staging_dir/bin/mipsel-linux-uclibc-gcc

acv, 2015-03-19:
This was needed back in the days to build packages. Is this still needed? Does it even work?

mhaas, 2015-03-19:
It is not needed on current Openwrt sdks. You can probably set $CC and be good.

acv, 2015-03-19:
Cool, I should get an openwrt SDK and play with it, last time I tried in the summer of '04 it was a pretty rough system.

mhaas, 2015-03-19:
With current OpenWRT SDKs, you need the Makefile (package recipe) we were talking about in that other ticket. You can probably just use their CC and make that work. With the Makefile, I only ever update the version and re-compile. That's tolerable for me, so I have no use for the current hack to directly use a cross-compiler.

acv, 2015-03-19:
Then PR should be made with:

1. This hack removed.
2. A readme type file with up to date and working build instructions.

PR #144 (2015-03-24): Remove ancient buildroot support from autogen.sh
This should resolve #103 if I understand right. Someone that actually uses build root and builds ipks should confirm (!)
closed 2015-03-24 by mhaas