wifidog/wifidog-gateway #107: create openwrt wifidog feed repo

champtar, 2015-03-18:
Hi,

can one of @acv, @benoitg, @florida63 create a new git repo like wifidog/old-openwrt-feed,
so we can put AA and BB openwrt makefile in it, with a quick howto, ...

Thanks in advance

florida63, 2015-03-18:
I can create a repository wifidog-feed if it's useful. 

No objection?

acv, 2015-03-18:
Go ahead. We can always remove it later if we need to.

sinkcup, 2015-03-18:
don't create for BB AND CC, should fork openwrt/packages

champtar, 2015-03-19:
@sinkcup, we still need a separate feed for AA and BB
i feel you don't understand how feed works

mhaas, 2015-03-19:
@champtar would it make sense to create a new repository (feed) which only contains the makefile for the old releases? That you won't have a feed which duplicates all packages for the old release.

The underlying assumption is that OpenWrt will only update the CC feed.

champtar, 2015-03-19:
@mhaas yes that's exactly what i'm trying to do with this issue.

mhaas, 2015-03-24:
FYI, wifidog and wifidog-tls are now building via https://github.com/wifidog/packages/tree/wifidog-experimental

florida63, 2015-03-24:
Good

mhaas, 2015-03-26:
@acv do you have the ability to create a new, empty repo for makefiles for the old openwrt releases?

florida63, 2015-03-26:
@mhass: I just create old-openwrt-feed.

databeille, 2015-03-27:
@acv, @mhass Guys, how do you plan to organize old-openwrt-feed ?

databeille, 2015-03-27:
@mhaas nice job for wifidog-tls :+1: ; I have successfully build a openWRT BB package, arch: ar71xx.

acv, 2015-03-27:
I never knew much about openwrt packaging, so I'm staying out of it. I only ever used wifidog on plain Linux and others in ISF (Benoit or Phil usually) did the openwrt stuff.

mhaas, 2015-03-27:
Well, I don't know.. I guess if you put the files into net/wifidog/ everything will be OK.

mhaas, 2015-03-27:
If you want different makefiles for different releases, then probably do different branches. At least the BB SDK supports checking out different branches conveniently.

closed 2020-11-14 by champtar