wifidog/wifidog-gateway #109: 1.2.0 a bit premature?

mhaas, 2015-03-19:
Hey there,

@sinkcup tagged and released 1.2.0 on Github. I think that's a bit premature, given that the changelog has not been fully updated. We should have some kind of consensus (or BDFL decision) on when to release. Now, we have people single-handedly tagging releases when they're not as ready as they could be.

florida63, 2015-03-19:
@sinkcup: I am of the same opinion as @mhaas.

In addition, before a tagged version 1.2.0 maybe would it make a pre-realease to test because there were many changes. 1.2.0_alpha for exemple.

mhaas, 2015-03-19:
Not sure if we need _alpha, everyone knows that x.y.0 releases are broken ;)

sinkcup, 2015-03-19:
It's a bit premature. maybe delete it is an good choice.
I will try to find a old commit ID, and tag it, such as 1.0.0, 1.1.5.
@florida63 If we use alpha, we should use 1.2.0-alpha.1 not 1.2.0_alpha
please see this docs http://semver.org/lang/zh-CN/ 
1.0.0-alpha < 1.0.0-alpha.1 < 1.0.0-beta < 1.0.0-beta.2 < 1.0.0-beta.11 < 1.0.0- rc.1 < 1.0.0

sinkcup, 2015-03-19:
I just want use tag as soon as possible, otherwise people use master, it's more unstable...
I will try to find the commit id of 1.1.5, and tag it.
You are right, should not release a tag by one person, need discussion

florida63, 2015-03-19:
1.1.4 --> https://github.com/wifidog/wifidog-gateway/commit/2f7b99ce80e831850baf82e689aa12919a6268df

mhaas, 2015-03-19:
I'll fix the changelog later today. I should have some time then.

sinkcup, 2015-03-19:
Sorry to trouble you.
I have delete 1.2.0, and add old tags.
I will become immersed in my work. See you.

acv, 2015-03-19:
This is a pretty good argument for having a 2 branch repo... With 2 branches master can be stable.

acv, 2015-03-19:
Do we need a BFDL? I mean I can do it if that's desired. Focus would be on the B and not so much the D ;-).

mhaas, 2015-03-19:
We don't need a dictator if everything happens as I command.. j/k. Just saying, there should be a consensus on "let's release" before we release.

databeille, 2015-03-19:
@mhaas I agree with you

mhaas, 2015-03-20:
Time for a release?

florida63, 2015-03-20:
This document (https://github.com/wifidog/wifidog-gateway/blob/master/NEWS) inconsistency on the layout. 

Otherwise ok to release 1.2.0

florida63, 2015-03-21:
For me it's ok.

mhaas, 2015-03-21:
Pinging @acv and @sinkcup. I think @sinkcup should do the release :)

florida63, 2015-03-21:
@champtar, @benoitg, @databeille. What about you?

sinkcup, 2015-03-21:
OK :+1: 
create 1.2.0 using commit 5cb3a3623533bc51ce2adf5590b7d9ef9b166302 ?

florida63, 2015-03-21:
@sinkcup:  yes

sinkcup, 2015-03-21:
@florida63 @mhaas  done :)

closed 2015-03-21 by sinkcup

florida63, 2015-03-21:
@mhaas: you might have to wait another

Perhaps, add the big changes comment in https://github.com/wifidog/wifidog-gateway/releases/tag/1.2.0

it lacks some useful files in your tar.gz (see old tar.gz we sourceforge)