wifidog/wifidog-gateway #117: Coverity: static code analysis

mhaas, 2015-03-21:
I went ahead and put my fork through the free coverity scan service.

There are some interesting defects uncovered: https://scan.coverity.com/projects/4579

You probably have to log in with github. If the consensus is that this kind of information/service is interesting, then I can work on getting the proper repo added to travis-ci and coverity.

florida63, 2015-03-21:
@mhaas:  good idea.

like this https://travis-ci.org/nodogsplash/nodogsplash/builds/54965717

in addition there is a link to each pull request. See red cross in https://github.com/nodogsplash/nodogsplash/pulls

mhaas, 2015-03-21:
I didn't know Travis tracked pull requests. That is most useful.

I wish it did track compiler warnings, though.

acv, 2015-03-21:
+1 on both ideas

acv, 2015-03-22:
I'm setting up travis-ci, or well, fumbling my way to having it build. Then we might look at hooking things like scan-build or coverity.

acv, 2015-03-22:
https://scan.coverity.com/projects/4595

For the actual wifidog/wifidog-gateway repo. I'll approve committers for access.

acv, 2015-03-22:
Apologies BTW for the semi functional travis-ci so far.

acv, 2015-03-22:
Thought it was my travis-ci config that was broken, turns out the build was actually broken... #127

acv, 2015-03-22:
Feels a bit like whack-a-mole. I've fixed a bunch of "high" issues and introduced new "high" issues (memory leaks...) in the way. That said, it's pretty outstanding as a static analyzer. Better than scan-build.

acv, 2015-03-24:
@mhaas, before adding features, I'll try to close out the remaining 27 coverity failures (and, undoubtedly, the 12 new ones I'll introduce while doing so!)

Feel free to go at it too, there's plenty to share.

#84 will feature in there due to how it plays fast and lose with the mutexes.

mhaas, 2015-03-24:
I will take care of some issues! Been busy lately with other stuff, but this is just too much fun.

acv, 2015-03-25:
I fixed about a dozen warnings from coverity tonight.

mhaas, 2015-03-25:
Same here - before the second coffee. Some threading issues remain which look false-positiveish to me.

mhaas, 2015-03-26:
We are hopefully done soon enough to get this in for 1.2.1.

acv, 2015-04-01:
Down to 3 issues.

acv, 2015-04-01:
I'll tackle the rewrite of the `auth.c` loop Soon <TM> based on #168 and that will take care of it (plus a trivial issue).