wifidog/wifidog-gateway #118: Arp_get function simplification?

florida63, 2015-03-21:
What do you think of using a command to simplify arp_get function (see firewall.c line 126) to get mac address.

eg "ip neigh show ip_adress -4 | cut -d -f5 '" returns directly mac

acv, 2015-03-21:
How often is `arp_get` invoked? If it's invoked rarely, that's OK. `fork()` + `execve()` is kinda slow if you're going to do it a lot.

Also, is ip present in OpenWRT?

acv, 2015-03-21:
Con for this:

* Current method works.

Pro for this:

* It'll be easier for anyone who wants to port the code to a non-Linux platform.

florida63, 2015-03-21:
Other positive point: 
If one day we adapt wifidog for ipv6, "/ proc / net / arp" is not compatible which leads me to this question.

acv, 2015-03-21:
I see.

`ip neigh show to IP` basically works. It's implemented with AF_NETLINK socket calls. I'd rather implement the lookup in wifidog itself instead of shelling out for performance reasons. Basically, the same reason we went with the current implementation: slow CPUs that don't context switch well in consumer routers.