wifidog/wifidog-gateway #119: Initialize cyassl only once

acv, 2015-03-21:
We should:

1. Extract the initialization of the `CYASSL_CTX` (and related cert loading) to a function.

2. Have the function cache the context in a static (file) global.

3. Have the function serialized through a mutex.

That way, initialization happens once and only once. basic pseudo-code:

    CYASSL_CTX *get_context(void):
        lock_mutex()
        if CYASSL_CTX != null:
            goto exit
        CYASSL_CTX = initialize()
    exit:
        ret = CYASSL_CTX
        unlock_mutex()
        return ret

mhaas, 2015-03-21:
+1

I should have done that from the start, but I needed it to work first!

acv, 2015-03-21:
I checked the CYASSL docs, this appears to all be thread safe if CYASSL_CTX creation is wrapped in a mutex.

champtar, 2015-03-21:
Hi all,
@acv, have you considered using libcurl? With libcurl you can use multiple SSL lib, it's already a depency of many other stuff i think, and it might be simpler to use than an SSL lib.

In any case thanks for your work.

mhaas, 2015-03-21:
@champtar libcurl (+cyassl) is bigger than cyassl alone. It might make sense to have libcurl to use polarssl, which is the default on OpenWrt. However, my OpenWrt image is running cyassl only, including uhttpd and key generation (via px5g-standalone). For normal images, you would probably end up with two SSL libs installed.

The original motivation for me:

1) keep my image as small as possible (below 4MB, for my router)
2) use a simple API

Regarding the second point, I looked at libcurl and did not find it any easier. Wifidog only speaks HTTP 1/0, so I only needed to change some socket handling. libcurl is not needed for that.

Ideas:
* do nothing :)
* provide libcurl as an alternative to cyassl
* switch OpenWrt over to CyaSSL instead of PolarSSL (because it works for me)

mhaas, 2015-03-22:
I'm closing this as the original issue is fixed. @champtar feel free to continue the discussion.

closed 2015-03-22 by mhaas