wifidog/wifidog-gateway #123: build error on openwrt buildroot after 6e1e36c650802c00105f03c4060af37a49991dd4

KazukiShimada, 2015-03-22:
Hi, all.
I keep on building wifidog on openwrt buildroot based on my feeds repo.
https://github.com/KazukiShimada/wifidog-openwrt/blob/master/wifidog/Makefile
After 6e1e36c650802c00105f03c4060af37a49991dd4, compile fails as follows,
---------------------------------------------------------------------------------------------------------
checking whether byte ordering is bigendian... (cached) yes
checking whether make sets $(MAKE)... (cached) yes
checking for ANSI C header files... (cached) yes
checking for doxygen... no
checking pthread.h usability... yes
checking pthread.h presence... yes
checking for pthread.h... yes
checking for pthread_create in -lpthread... yes
Begining libhttpd dependencies check
checking for string.h... (cached) yes
checking for strings.h... (cached) yes
checking stdarg.h usability... yes
checking stdarg.h presence... yes
checking for stdarg.h... yes
checking for unistd.h... (cached) yes
checking for main in -lsocket... no
checking for main in -lnsl... yes
libhttpd dependencies check complete
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: error: cannot find input file: `Makefile.in'
make[3]: *** [/home/zukky/openwrt/trunk/build_dir/target-mips_mips32_uClibc-0.9.33.2/wifidog-20150321/.configured_] Error 1
make[3]: Leaving directory `/home/zukky/openwrt/trunk/feeds/wifidog/wifidog'
make[2]: *** [package/feeds/wifidog/wifidog/compile] Error 2
make[2]: Leaving directory `/home/zukky/openwrt/trunk'
make[1]: *** [/home/zukky/openwrt/trunk/staging_dir/target-mips_mips32_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/zukky/openwrt/trunk'
make: *** [world] エラー 2
zukky@ubuntu:~/openwrt/trunk$ 
---------------------------------------------------------------------------------------------------------
It build successfully previous revision 304eb68f2c3330f5ddd26feca5ca8658ef17119a .
I confirmed it latest buildroot and mips arch.
Any advice is appreciated.

Thanks in advance.

acv, 2015-03-22:
Can you do a brand new check out of the wifidog code and then run autogen,sh on it?

acv, 2015-03-22:
I had similar problems on Ubuntu because `README` got renamed `README.md` and it confused `automake`. Re-running `autogen.sh` fixed it for me.

acv, 2015-03-22:
Indeed, the build was broken. GNU coding standards require a README file and deviations are rewarded with autotools failures because :gnu: I have updated configs and the HEAD of master (c9fdc549dc7465ba5c63917a021f003f1609b11f) should build for you now.

acv, 2015-03-22:
30fcec7e9e77c2472da4460cfd795c718b2b79e0 is confirmed to be able to be built from a clean clone. Sorry for the trouble, we changed text files (README and INSTALL) and it broke automake ;-(

On dev check outs, the build could still run if the previous README and INSTALL were present so it avoided detection.

We've added travis-ci to spot these issues.

Thanks for reporting this issue.

florida63, 2015-03-22:
with the last release (https://github.com/wifidog/wifidog-gateway/commit/5fae154ff41f328426729bb6ebefde3df1cb6a55)

root@Auz2:/etc/init.d# wifidog -f -d 7
/usr/bin/wifidog: 1: Syntax error: word unexpected (expecting ")")

mhaas, 2015-03-22:
@florida63 did you built straight from source or with a package makefile? The makefile might have installed the init script instead of /usr/bin/wifidog

florida63, 2015-03-22:
with a makefile. /usr/bin/wifidog has been installed.

mhaas, 2015-03-22:
Can you post the makefile?

KazukiShimada, 2015-03-22:
@acv 
I confirmed it build successfully at latest revision.
Thank you for your quick fix.

florida63, 2015-03-22:
@mmhas http://pastebin.com/85Jud443

it compile under Whiterussian. I know this is not new but I have several router that runs over and lack of time I have not adapted the specific firmware. faster for me to update the ipk.

florida63, 2015-03-22:
It's ok with https://github.com/wifidog/wifidog-gateway/commit/72f3d3d4ec157d0988da64adeda5f64f8f2695d8

We can close this issue.

mhaas, 2015-03-22:
@florida63 cool, thanks

closed 2015-03-22 by mhaas