wifidog/wifidog-gateway #138: Integrate coverity with travis

mhaas, 2015-03-22:
It's quite straightforward: https://scan.coverity.com/travis_ci

Once set up, we can trigger coverity through commits to a special branch.

Assigning to @acv as he is the master of coverity.

acv, 2015-03-22:
One PITA is that coverity will run with and without SSL on both gcc and clang.

acv, 2015-03-22:
Side effect seems to be that you always pass since it doesn't find any different number of issues between build the last two builds it runs.

acv, 2015-03-22:
I could make the branch only run gcc + SSL by keeping its .travis.yml distinct. We just gotta remember to fix it when/if we push to it a new .travis.yml file.

mhaas, 2015-03-22:
Sounds good. I wonder if we can limit the coverity run to only one entry in the build matrix. Seems like what you are suggesting is the current best practice :/

https://github.com/travis-ci/travis-ci/issues/1975

acv, 2015-03-22:
> travis-ci/travis-ci#1975

Yeah, that's where I got the idea...

acv, 2015-03-22:
So I updated the coverity_scan branch and travis appears to be doing "the right thing" and only doing gcc+SSL.

mhaas, 2015-03-22:
Awesome.

closed 2015-03-22 by mhaas