wifidog/wifidog-gateway #145: Mac Trusted dupliquer si present plusieurs fois dans la liste

florida63, 2015-03-22:
If we capture in wifidog.conf

TrustedMACList 00:b0:D0:FE:7C:2A,00:b0:D0:FE:7C:2A,00:b0:D0:FE:7C:2A

obtained:
root@Auz2:/etc/init.d# wdctl status
WiFiDog status

Version: 1.2.0
Uptime: 0d 0h 0m 9s
Has been restarted: no
Internet Connectivity: yes
Auth server reachable: no
Clients served this session: 0

0 clients connected.

Trusted MAC addresses:
  00:b0:D0:FE:7C:2A
  00:b0:D0:FE:7C:2A
  00:b0:D0:FE:7C:2A

There are no duplicate check

acv, 2015-03-22:
Since we already seek to the end of the list to insert, we might as well have it check as it scans the list.

florida63, 2015-03-22:
thanks for title

acv, 2015-03-22:
Incidentally, most of the config makes no unique checks.

acv, 2015-03-22:
Plaisir ;-)

acv, 2015-03-22:
The fix is in.

closed 2015-03-22 by acv