wifidog/wifidog-gateway #148: wdctl status very slow return when auth is not accessible.

florida63, 2015-03-23:
when auth is not accessible wdctl status very slow return

when there is an answer
root@Auz2:/etc# wdctl status
[3][Sat Jan  1 03:47:13 2000][5165](centralserver.c:148) Failed to connect to any of the auth servers
\��*\��*.1WiFiDog status

Version: 1.2.1
Uptime: 0d 0h 4m 24s
Has been restarted: yes (from PID 4381)
Internet Connectivity: yes
Auth server reachable: no
Clients served this session:

florida63, 2015-03-23:
@acv: when I make several wdctl status I get a Segmentation fault

[7][Sat Jan  1 13:04:39 2000][16632](ping_thread.c:199) Auth Server Says: Pong
[7][Sat Jan  1 13:04:52 2000][16631](wdctl_thread.c:139) Accepted connection on wdctl socket 4367976 ()
[7][Sat Jan  1 13:04:52 2000][16668](wdctl_thread.c:162) Entering thread_wdctl_handler....
[7][Sat Jan  1 13:04:52 2000][16668](wdctl_thread.c:166) Read bytes and stuff from 9
[7][Sat Jan  1 13:04:52 2000][16668](wdctl_thread.c:196) Request received: [status]
[7][Sat Jan  1 13:04:52 2000][16668](util.c:408) Locking client list
[7][Sat Jan  1 13:04:52 2000][16668](util.c:408) Client list locked
[7][Sat Jan  1 13:04:52 2000][16668](util.c:437) Unlocking client list
[7][Sat Jan  1 13:04:52 2000][16668](util.c:437) Client list unlocked
[7][Sat Jan  1 13:04:52 2000][16668](util.c:451) Locking config
[7][Sat Jan  1 13:04:52 2000][16668](util.c:451) Config locked
[7][Sat Jan  1 13:04:52 2000][16668](util.c:457) Unlocking config
[7][Sat Jan  1 13:04:52 2000][16668](util.c:457) Config unlocked
[7][Sat Jan  1 13:04:52 2000][16668](wdctl_thread.c:212) Exiting thread_wdctl_handler....
[7][Sat Jan  1 13:05:15 2000][16631](wdctl_thread.c:139) Accepted connection on wdctl socket 4367976 ()
[7][Sat Jan  1 13:05:15 2000][16674](wdctl_thread.c:162) Entering thread_wdctl_handler....
[7][Sat Jan  1 13:05:15 2000][16674](wdctl_thread.c:166) Read bytes and stuff from 9
[7][Sat Jan  1 13:05:15 2000][16674](wdctl_thread.c:196) Request received: [status]
[7][Sat Jan  1 13:05:15 2000][16674](util.c:408) Locking client list
[7][Sat Jan  1 13:05:15 2000][16674](util.c:408) Client list locked
[7][Sat Jan  1 13:05:15 2000][16674](util.c:437) Unlocking client list
[7][Sat Jan  1 13:05:15 2000][16674](util.c:437) Client list unlocked
[7][Sat Jan  1 13:05:15 2000][16674](util.c:451) Locking config
[7][Sat Jan  1 13:05:15 2000][16674](util.c:451) Config locked
[7][Sat Jan  1 13:05:15 2000][16674](util.c:457) Unlocking config
[7][Sat Jan  1 13:05:15 2000][16674](util.c:457) Config unlocked
[7][Sat Jan  1 13:05:15 2000][16674](wdctl_thread.c:212) Exiting thread_wdctl_handler....
[7][Sat Jan  1 13:05:16 2000][16631](wdctl_thread.c:139) Accepted connection on wdctl socket 4367976 ()
[7][Sat Jan  1 13:05:16 2000][16676](wdctl_thread.c:162) Entering thread_wdctl_handler....
[7][Sat Jan  1 13:05:16 2000][16676](wdctl_thread.c:166) Read bytes and stuff from 9
[7][Sat Jan  1 13:05:16 2000][16676](wdctl_thread.c:196) Request received: [status]
[7][Sat Jan  1 13:05:16 2000][16676](util.c:408) Locking client list
[7][Sat Jan  1 13:05:16 2000][16676](util.c:408) Client list locked
[7][Sat Jan  1 13:05:16 2000][16676](util.c:437) Unlocking client list
[7][Sat Jan  1 13:05:16 2000][16676](util.c:437) Client list unlocked
[7][Sat Jan  1 13:05:16 2000][16676](util.c:451) Locking config
[7][Sat Jan  1 13:05:16 2000][16676](util.c:451) Config locked
[7][Sat Jan  1 13:05:16 2000][16676](util.c:457) Unlocking config
[7][Sat Jan  1 13:05:16 2000][16676](util.c:457) Config unlocked
[7][Sat Jan  1 13:05:16 2000][16676](wdctl_thread.c:212) Exiting thread_wdctl_handler....
[7][Sat Jan  1 13:05:17 2000][16631](wdctl_thread.c:139) Accepted connection on wdctl socket 4367992 ()
[7][Sat Jan  1 13:05:17 2000][16678](wdctl_thread.c:162) Entering thread_wdctl_handler....
[7][Sat Jan  1 13:05:17 2000][16678](wdctl_thread.c:166) Read bytes and stuff from 9
[7][Sat Jan  1 13:05:17 2000][16678](wdctl_thread.c:196) Request received: [status]
[7][Sat Jan  1 13:05:17 2000][16678](util.c:408) Locking client list
[7][Sat Jan  1 13:05:17 2000][16678](util.c:408) Client list locked
[7][Sat Jan  1 13:05:17 2000][16678](util.c:437) Unlocking client list
[7][Sat Jan  1 13:05:17 2000][16678](util.c:437) Client list unlocked
[7][Sat Jan  1 13:05:17 2000][16678](util.c:451) Locking config
[7][Sat Jan  1 13:05:17 2000][16678](util.c:451) Config locked
[7][Sat Jan  1 13:05:17 2000][16678](util.c:457) Unlocking config
[7][Sat Jan  1 13:05:17 2000][16678](util.c:457) Config unlocked
[7][Sat Jan  1 13:05:17 2000][16678](wdctl_thread.c:212) Exiting thread_wdctl_handler....
[7][Sat Jan  1 13:05:18 2000][16631](wdctl_thread.c:139) Accepted connection on wdctl socket 4367976 ()
[7][Sat Jan  1 13:05:18 2000][16680](wdctl_thread.c:162) Entering thread_wdctl_handler....
[7][Sat Jan  1 13:05:18 2000][16680](wdctl_thread.c:166) Read bytes and stuff from 9
[7][Sat Jan  1 13:05:18 2000][16680](wdctl_thread.c:196) Request received: [status]
[7][Sat Jan  1 13:05:18 2000][16680](util.c:408) Locking client list
[7][Sat Jan  1 13:05:18 2000][16680](util.c:408) Client list locked
[7][Sat Jan  1 13:05:18 2000][16680](util.c:437) Unlocking client list
[7][Sat Jan  1 13:05:18 2000][16680](util.c:437) Client list unlocked
[7][Sat Jan  1 13:05:18 2000][16680](util.c:451) Locking config
[7][Sat Jan  1 13:05:18 2000][16680](util.c:451) Config locked
[7][Sat Jan  1 13:05:18 2000][16680](util.c:457) Unlocking config
[7][Sat Jan  1 13:05:18 2000][16680](util.c:457) Config unlocked
[7][Sat Jan  1 13:05:18 2000][16680](wdctl_thread.c:212) Exiting thread_wdctl_handler....
[7][Sat Jan  1 13:05:18 2000][16631](wdctl_thread.c:139) Accepted connection on wdctl socket 4367976 ()
[7][Sat Jan  1 13:05:18 2000][16682](wdctl_thread.c:162) Entering thread_wdctl_handler....
[... 95 lines cut ...]
[7][Sat Jan  1 13:05:22 2000][16698](util.c:437) Unlocking client list
[7][Sat Jan  1 13:05:22 2000][16698](util.c:437) Client list unlocked
[... cut ...]
[7][Sat Jan  1 13:05:23 2000][16700](util.c:457) Config unlocked
[7][Sat Jan  1 13:05:23 2000][16700](wdctl_thread.c:212) Exiting thread_wdctl_handler....
[7][Sat Jan  1 13:05:26 2000][16631](wdctl_thread.c:139) Accepted connection on wdctl socket 4367992 ()
[7][Sat Jan  1 13:05:26 2000][16702](wdctl_thread.c:162) Entering thread_wdctl_handler....
[7][Sat Jan  1 13:05:26 2000][16702](wdctl_thread.c:166) Read bytes and stuff from 9
[7][Sat Jan  1 13:05:26 2000][16702](wdctl_thread.c:196) Request received: [status]
Segmentation fault

acv, 2015-03-23:
Awesome... Sounds like a race somewhere.

acv, 2015-03-23:
Can you repeat on a PC and get a core file?

acv, 2015-03-23:
Also, how far along did it get in printing the status?

florida63, 2015-03-23:
The status is full every time, and it crashes randomly when I do wdctl status.

root@Auz2:~# wdctl status
wdctl: wifidog probably not started (Error: Connection refused)
root@Auz2:~# wdctl status

>Can you repeat on a PC and get a core file?
I tested on a PC and I can't simulated the problem.

acv, 2015-03-24:
I'm trying 3 threads running wdctl status in a tight loop 20000 times to try and reproduce the race.

acv, 2015-03-24:
Ok. So the delay as to do with the Mutex on the auth server list. It's held by the thread that is checking the auth server for availability. I'm going to configure a timeout there I think.

acv, 2015-03-24:
> [3][Sat Jan 1 03:47:13 2000]5165 Failed to connect to any of the auth servers
\��\��.1

That's not from wdctl. It might be from wifidog in the background. Can you try with wifidog running in a different shell so the two output stream don't overlap?

acv, 2015-03-24:
So timeout don't work on `connect()` and `wdctl` is the only runtime use of that lock. I think we'll live with slow `wdctl`.

acv, 2015-03-24:
I get this:

```
xWiFiDog status

Version: 1.2.1
Uptime: 0d 0h 2m 37s
Has been restarted: yes (from PID 29780)
Internet Connectivity: yes
Auth server reachable: no
Clients served this session: 0
```

That x is weird...

And it goes away if I need the buffer for a new pstr_t.

acv, 2015-03-24:
Depending on heap allocator behaviour, this can *also* explain the segfault.

acv, 2015-03-24:
@florida63 Can you try to reproduce with my PR #149? I may have it resolved.

Also, when using freshly malloc'd memory for string operations, always make sure it contains a NULL or 3...

florida63, 2015-03-24:
I look at it in the day

mhaas, 2015-03-24:
Maybe we want to put the new pstring code through coverity.

florida63, 2015-03-24:
@acv: It seems correct the two problems. (\��\�� and Segmentation fault).

florida63, 2015-03-24:
I specify that I launched a hundred times without wdctl crash.

In addition, the return seems faster.

"memset (new-> buf, 0, MAX_BUF);" seems to be very beneficial.

mhaas, 2015-03-24:
As I am not very good with C, I google the issue and found this:

https://stackoverflow.com/questions/13410550/do-i-have-to-memset-after-malloc-new-memory

Perhaps we should audit every call to malloc and make surE:
a) memset is called OR
b) for strings, that \0 is inserted

acv, 2015-03-24:
@mhaas, I was thinking just that. Since we use `safe_malloc()` everywhere, I thought we could just put the `memset` in the `safe_malloc` then audit the code to remove double `memset`.

Seems the smartest way to go about it.

acv, 2015-03-24:
Going to close this specific issue and open a new one to refactor `safe_malloc`.

closed 2015-03-24 by acv

mhaas, 2015-03-24:
good idea.

acv, 2015-03-24:
@mhaas Re-ran coverity on the code base. I find it easier to do manually then use the github integration tbh.

mhaas, 2015-03-24:
@acv it's still good to have the cloud-based trigger because I often switch computers. Easier for me, so it's good to have both methods.