wifidog/wifidog-gateway #151: wifidog could not start Automatically

julor, 2015-03-24:
openwrt version:OpenWrt Barrier Breaker 14.07 / LuCI Trunk (0.12+svn-r10530)
I installed wifidog correctly,it could start manually by useing '/etc/init.d/wifidog start'.
I have used '/etc/init.d/wifidog enable',and found S65wifidog in '/etc/rc.d/',while reboot,wifidog isn't started yet.
could anyone help me?

mhaas, 2015-03-24:
I have seen similar behavior and I attributed this to the interface not being up. I wrote a script to trigger a wifidog reload on ifup.

Oddly enough, with current master the problem is gone. There had been some changes - now wifidog waits for the interface to come up. What version are you running? Can you build your own package?

julor, 2015-03-24:
I could build wifidog using openwrt-sdk(Barrier Breaker  14.07)
wifidog is the laster version on master

mhaas, 2015-03-24:
Hm, that's weird. What device? What Makefile/init script did you use?

julor, 2015-03-24:
Makefile is openwrt-sdk default settings. 

device:TP-Link TL-WR841N/ND v7

wifidog.init:
#!/bin/sh /etc/rc.common
# Copyright (C) 2006 OpenWrt.org
START=65
EXTRA_COMMANDS="status"
EXTRA_HELP="        status Print the status of the service"

start() {
	/usr/bin/wifidog-init start
}

stop() {
	/usr/bin/wifidog-init stop
}

status() {
	/usr/bin/wifidog-init status
}

mhaas, 2015-03-24:
Can you try this one:
http://pastebin.ca/2963126

julor, 2015-03-24:
oh,it works!
thank you

mhaas, 2015-03-24:
Let me re-open this - I wonder if we have to fix the underlying issue in Wifidog. I think the new init script only works because procd restarts wifidog.

databeille, 2015-04-21:
What do you mean by "underlying" ?