wifidog/wifidog-gateway #157: Time for a release 1.2.1?

florida63, 2015-03-24:
The latest patches require a realease

acv, 2015-03-24:
Sure, update NEWS and cut the release.

mhaas, 2015-03-24:
Didn't someone report a segfault a couple of hours ago?

acv, 2015-03-24:
1.2.0 has a known segfault in `src/util.c:get_status_text()` that would be fixed by a 1.2.1.

mhaas, 2015-03-24:
Ok. Then go for it.

florida63, 2015-03-24:
update https://github.com/wifidog/wifidog-gateway/releases/tag/1.2.0

mhaas, 2015-03-24:
1.2.0 does not even build ;) good thinking on the update.

acv, 2015-04-01:
I say now is a great time.

sinkcup, 2015-04-16:
...
We have made a mistake about git flow. master should ready for release at all times.
E.G.
if the next mileston after 1.2.0 is 1.3.0, if there is a urgent bug, we should release 1.2.1 from master.
if the next mileston after 1.2.0 is 1.2.1, if there is a urgent bug, we should move all issues to 1.2.2, then fix it and release 1.2.1 from master.

mhaas, 2015-04-16:
Master is absolutely ready for a release! We have had only fixes and one very minor feature.

sinkcup, 2015-04-16:
I know, code had been tested before merge into master.
I means can we move other issues, and release 1.2.1 now?
The best time for releasing 1.2.1 is 15 days ago, secondly it's now.

mhaas, 2015-04-16:
Sure! I think nobody ever felt like releasing .1, but it's time!

sinkcup, 2015-04-16:
OK, I do it now.

sinkcup, 2015-04-16:
I see the `NEWS` and the version in `configure.in` are correct, so I have released 1.2.1.
Is it right? @acv

mhaas, 2015-04-16:
I am not quite sure if we have everything in NEWS, e.g. Some refactoring ie missing. But it's mostly complete.

closed 2015-04-16 by sinkcup

acv, 2015-04-16:
I don't know if refactoring that don't affect functionality really belong in a NEWS files anyway. If they do, it's as a single line.

mhaas, 2015-04-17:
This is a very good release, by the way. Lots of polishing!

[image]