wifidog/wifidog-gateway #171: Proxy: Inactive authentication if browser is configured with the proxy parameter.

florida63, 2015-03-31:
It is necessary to add a rule avoids this problem.

acv, 2015-03-31:
Can you clarify what you mean? I'm having a hard time following.

databeille, 2015-03-31:
It I understand well, it seems related to the port used by the proxy.
Would you like to catch/redirect all ports to 2060 and not only 80 in iptables ?

acv, 2015-03-31:
The downside in catching all ports is it will catch a lot of non-HTTP.

Port is already configurable. Maybe that can be extended to allow for the configuration of one, many, ranges or all port?

mhaas, 2015-03-31:
Assuming you want to catch browsers which use a proxy, wouldn't that require that wifidog speaks the socks protocol?

florida63, 2015-03-31:
@acv : In fact, if you have configured proxyport with tinyproxy you can surf without you authenticated. For this, you only have to parameterize the browser with the IP gateway and proxy port.

To correct it, block the proxy port for the IP gateway.

florida63, 2015-04-02:
I manually tested these 2 rules and it seems to fix the problem.

iptables -I INPUT 1 -i br0 -m mark --mark 0x2 -p tcp --dport 8888 -j ACCEPT
iptables -I INPUT 1 -i br0 -p tcp --dport 8888 -j REJECT