wifidog/wifidog-gateway #175: Some valgrind errors

mhaas, 2015-04-01:
==23879== Memcheck, a memory error detector
==23879== Copyright (C) 2002-2013, and GNU GPL'd, by Julian Seward et al.
==23879== Using Valgrind-3.10.0 and LibVEX; rerun with -h for copyright info
==23879== Command: ../../src/.libs/wifidog -d 7 -f -c wifidog-mock.conf -a /tmp/arp
==23879== Parent PID: 22565
==23879== 
==23879== Thread 8:
==23879== Conditional jump or move depends on uninitialised value(s)
==23879==    at 0x8051B28: iptables_do_command (fw_iptables.c:112)
==23879==    by 0x8052D55: iptables_fw_access (fw_iptables.c:569)
==23879==    by 0x80507BC: _fw_deny_raw (firewall.c:138)
==23879==    by 0x80507BC: fw_deny (firewall.c:125)
==23879==    by 0x804B88C: http_callback_auth (http.c:282)
==23879==    by 0x403BCBA: httpdProcessRequest (api.c:836)
==23879==    by 0x804E304: thread_httpd (httpd_thread.c:67)
==23879==    by 0x405BF15: start_thread (pthread_create.c:309)
==23879==    by 0x415D11D: clone (clone.S:129)
==23879== 
==23879== Use of uninitialised value of size 4
==23879==    at 0x40B50FB: _itoa_word (_itoa.c:179)
==23879==    by 0x40B8808: vfprintf (vfprintf.c:1641)
==23879==    by 0x40BA57A: buffered_vfprintf (vfprintf.c:2335)
==23879==    by 0x40B58C4: vfprintf (vfprintf.c:1296)
==23879==    by 0x416C047: __vfprintf_chk (vfprintf_chk.c:33)
==23879==    by 0x80505F5: vfprintf (stdio2.h:127)
==23879==    by 0x80505F5: _debug (debug.c:59)
==23879==    by 0x8051B4C: iptables_do_command (fw_iptables.c:115)
==23879==    by 0x8052D55: iptables_fw_access (fw_iptables.c:569)
==23879==    by 0x80507BC: _fw_deny_raw (firewall.c:138)
==23879==    by 0x80507BC: fw_deny (firewall.c:125)
==23879==    by 0x804B88C: http_callback_auth (http.c:282)
==23879==    by 0x403BCBA: httpdProcessRequest (api.c:836)
==23879==    by 0x804E304: thread_httpd (httpd_thread.c:67)
==23879== 
==23879== Conditional jump or move depends on uninitialised value(s)
==23879==    at 0x40B5103: _itoa_word (_itoa.c:179)
==23879==    by 0x40B8808: vfprintf (vfprintf.c:1641)
==23879==    by 0x40BA57A: buffered_vfprintf (vfprintf.c:2335)
==23879==    by 0x40B58C4: vfprintf (vfprintf.c:1296)
==23879==    by 0x416C047: __vfprintf_chk (vfprintf_chk.c:33)
==23879==    by 0x80505F5: vfprintf (stdio2.h:127)
==23879==    by 0x80505F5: _debug (debug.c:59)
==23879==    by 0x8051B4C: iptables_do_command (fw_iptables.c:115)
==23879==    by 0x8052D55: iptables_fw_access (fw_iptables.c:569)
==23879==    by 0x80507BC: _fw_deny_raw (firewall.c:138)
==23879==    by 0x80507BC: fw_deny (firewall.c:125)
==23879==    by 0x804B88C: http_callback_auth (http.c:282)
==23879==    by 0x403BCBA: httpdProcessRequest (api.c:836)
==23879==    by 0x804E304: thread_httpd (httpd_thread.c:67)

==23879== Memcheck, a memory error detector
==23879== Copyright (C) 2002-2013, and GNU GPL'd, by Julian Seward et al.
==23879== Using Valgrind-3.10.0 and LibVEX; rerun with -h for copyright info
==23879== Command: ../../src/.libs/wifidog -d 7 -f -c wifidog-mock.conf -a /tmp/arp
==23879== Parent PID: 22565
==23879== 
==23879== Thread 8:
==23879== Conditional jump or move depends on uninitialised value(s)
==23879==    at 0x8051B28: iptables_do_command (fw_iptables.c:112)
==23879==    by 0x8052D55: iptables_fw_access (fw_iptables.c:569)
==23879==    by 0x80507BC: _fw_deny_raw (firewall.c:138)
==23879==    by 0x80507BC: fw_deny (firewall.c:125)
==23879==    by 0x804B88C: http_callback_auth (http.c:282)
==23879==    by 0x403BCBA: httpdProcessRequest (api.c:836)
==23879==    by 0x804E304: thread_httpd (httpd_thread.c:67)
==23879==    by 0x405BF15: start_thread (pthread_create.c:309)
==23879==    by 0x415D11D: clone (clone.S:129)
==23879== 
==23879== Use of uninitialised value of size 4
==23879==    at 0x40B50FB: _itoa_word (_itoa.c:179)
==23879==    by 0x40B8808: vfprintf (vfprintf.c:1641)
==23879==    by 0x40BA57A: buffered_vfprintf (vfprintf.c:2335)
==23879==    by 0x40B58C4: vfprintf (vfprintf.c:1296)
==23879==    by 0x416C047: __vfprintf_chk (vfprintf_chk.c:33)
==23879==    by 0x80505F5: vfprintf (stdio2.h:127)
==23879==    by 0x80505F5: _debug (debug.c:59)
==23879==    by 0x8051B4C: iptables_do_command (fw_iptables.c:115)
==23879==    by 0x8052D55: iptables_fw_access (fw_iptables.c:569)
==23879==    by 0x80507BC: _fw_deny_raw (firewall.c:138)
==23879==    by 0x80507BC: fw_deny (firewall.c:125)
==23879==    by 0x804B88C: http_callback_auth (http.c:282)
==23879==    by 0x403BCBA: httpdProcessRequest (api.c:836)
==23879==    by 0x804E304: thread_httpd (httpd_thread.c:67)
==23879== 
==23879== Conditional jump or move depends on uninitialised value(s)
==23879==    at 0x40B5103: _itoa_word (_itoa.c:179)
==23879==    by 0x40B8808: vfprintf (vfprintf.c:1641)
==23879==    by 0x40BA57A: buffered_vfprintf (vfprintf.c:2335)
==23879==    by 0x40B58C4: vfprintf (vfprintf.c:1296)
==23879==    by 0x416C047: __vfprintf_chk (vfprintf_chk.c:33)
==23879==    by 0x80505F5: vfprintf (stdio2.h:127)
==23879==    by 0x80505F5: _debug (debug.c:59)
==23879==    by 0x8051B4C: iptables_do_command (fw_iptables.c:115)
==23879==    by 0x8052D55: iptables_fw_access (fw_iptables.c:569)
==23879==    by 0x80507BC: _fw_deny_raw (firewall.c:138)
==23879==    by 0x80507BC: fw_deny (firewall.c:125)
==23879==    by 0x804B88C: http_callback_auth (http.c:282)
==23879==    by 0x403BCBA: httpdProcessRequest (api.c:836)
==23879==    by 0x804E304: thread_httpd (httpd_thread.c:67)

==23879== Thread 7:
==23879== Conditional jump or move depends on uninitialised value(s)
==23879==    at 0x8051B28: iptables_do_command (fw_iptables.c:112)
==23879==    by 0x8052D31: iptables_fw_access (fw_iptables.c:563)
==23879==    by 0x8050710: fw_allow (firewall.c:87)
==23879==    by 0x804BEE3: authenticate_client (auth.c:192)
==23879==    by 0x804BA60: http_callback_auth (http.c:306)
==23879==    by 0x403BCBA: httpdProcessRequest (api.c:836)
==23879==    by 0x804E304: thread_httpd (httpd_thread.c:67)
==23879==    by 0x405BF15: start_thread (pthread_create.c:309)
==23879==    by 0x415D11D: clone (clone.S:129)
==23879== 
==23879==

mhaas, 2015-04-01:
Note, may come from -O1

acv, 2015-04-02:
Ultimately, we were *probably* reading initialized memory because of how `waitpid()` works. (I can't reproduce but 90% convinced it should be fixed, maybe ;-))

mhaas, 2015-04-02:
Yeah, I would often see -1 as exit value. OTOH, I did not see the error again with -O1.