wifidog/wifidog-gateway #191: Can't get userAgent

eggbucket, 2015-04-11:
httpd.h
typedef struct {
        int method, contentLength, authLength;
        char path[HTTP_MAX_URL], query[HTTP_MAX_URL],    host[HTTP_MAX_URL],ifModified[HTTP_MAX_URL]; /* acv@acv.ca/wifidog: Added decoding
                                                                                   of host: header if present. */
        
	char userAgent[HTTP_MAX_URL];
	char referer[HTTP_MAX_URL];
	char contentType[HTTP_MAX_URL];

	char authUser[HTTP_MAX_AUTH];
	char authPassword[HTTP_MAX_AUTH];
    } httpReq;

debug(LOG_DEBUG, "userAgent=%s",r->request.userAgent);

output     userAgent=

acv, 2015-04-13:
The code that parses userAgent was removed from libhttpd because it contained buffer overflows and continuously crashed gateways.

acv, 2015-04-13:
Same for Referer and Content-Type (in the request, no response).

acv, 2015-04-13:
If you use the current master, the def should be:

```C
    typedef struct {
        int method, contentLength, authLength;
        char path[HTTP_MAX_URL], query[HTTP_MAX_URL], host[HTTP_MAX_URL],       /* acv@acv.ca/wifidog: Added decoding
                                                                                   of host: header if present. */
         ifModified[HTTP_MAX_URL];
        char authUser[HTTP_MAX_AUTH];
        char authPassword[HTTP_MAX_AUTH];
    } httpReq;
```