wifidog/wifidog-gateway #192: Could you help me on the traffic statistics?

t123yh, 2015-04-12:
I'm creating a hotspot that charges by the user's traffic. I've been writting an web authentication system based on ASP.NET MVC, but I found that I cannot count the traffic on each /auth/ action because the incoming and outgoing value won't be reset to zero each report. Could you please tell me where can I modify the code that it can let the traffic statistic be automatically reseted to zero? Or could you please add a section to the configuration file that controls it? Thanks a lot!

acv, 2015-04-13:
In a way, that'd be unfair because if there's no traffic, the gateway will create some (it pings the user.)

That said, if you keep track of the previous value sent in the connection, you can sub from it from new values and get only the delta.

Adding delta stats is possible, the gateway keeps around most of the required information. The code that sends the stats is in `src/firewall.c` at:

https://github.com/wifidog/wifidog-gateway/blob/master/src/firewall.c#L287-L290

You would need to add two new counters to the `t_counters` type (https://github.com/wifidog/wifidog-gateway/blob/master/src/client_list.h#L36-L42) and also change the counters update code in:

https://github.com/wifidog/wifidog-gateway/blob/master/src/fw_iptables.c#L628-L725

acv, 2015-04-13:
A patch would be considered for merging if it did that only optionally based on a config file directive.

t123yh, 2015-04-14:
Awesome! I can't thank you anymore.