wifidog/wifidog-gateway #193: No web site sent to client on logout

mhaas, 2015-04-21:
If a client requests a logout, the socket is closed. Instead, Wifidog should send a "logout successful" page or at least HTTP 200 with an empty body.

The offending code is in http.c, around line 270:
```
   if ((client = client_list_find(r->clientAddr, mac)) == NULL) {
                debug(LOG_DEBUG, "New client for %s", r->clientAddr);
                client_list_add(r->clientAddr, mac, token->value);
            } else if (logout) {
                logout_client(client);
            } else {
                debug(LOG_DEBUG, "Client for %s is already in the client list", client->ip);
            }
```

gareth41, 2015-08-08:
Try changing to this and it will redirect back to portal after manual logout, you can set custom url parameters depending on your auth server.

            if ((client = client_list_find(r->clientAddr, mac)) == NULL) {
                debug(LOG_DEBUG, "New client for %s", r->clientAddr);
                client_list_add(r->clientAddr, mac, token->value);
            } else if (logout) {
                logout_client(client);
                t_auth_serv *auth_server = get_auth_server();
                char *urlFragment;
                safe_asprintf(&urlFragment, "%sauth=logout", auth_server->authserv_login_script_path_fragment);
                debug(LOG_INFO, "Logout requested by client %s mac %s", r->clientAddr, mac);
                http_send_redirect_to_auth(r, urlFragment, "Redirect to portal after logout");
                free(urlFragment);
            } else {
                debug(LOG_DEBUG, "Client for %s is already in the client list", client->ip);
            }

t123yh, 2015-08-11:
I think you can redirect the client to the auth server.

benoitg, 2015-08-14:
@gareth41 If you really want to do this, the URL to redirect to should come from the auth server (retrieved in logout_client), not the other way around.

But honestly the local logout route was more a debugging tool than anything else.