wifidog/wifidog-gateway #197: how to add traffic control

strong46066999, 2015-04-23:
use ifb or imq?

acv, 2015-04-24:
I believe this is orthogonal to WifiDog and should be down at the underlying O/S level. That said I don't really know much about traffic shaping / QoS on Linux in general.

A major limitations for this is that wifidog is compatible with some pretty old platforms (i.e.: RHEL/CentOS down to like RHEL3 and OpenWRT versions from the last 10 years.)

acv, 2015-04-24:
Maybe we could add hooks at various places in the firewall processing code to shell out to some shell script and allow you to arbitrarily hook up ifb, imq or whatever else you want?

closed 2015-05-03 by acv

t123yh, 2015-05-13:
I think in the further versions we may execute scripts transmitted from server(or script in local and parameters from server) in order to modify some QoS rules for each user.