wifidog/wifidog-gateway #208: md checksum to stop tampering with auth/redirect query strings

gareth41, 2015-05-11:
There should be a checksum parameter attached to the auth query string and the login/redirect query string too, this will stop any tampering with the parameters and bogus info being sent to the auth server.  coova-chilli has this, the uamsecret is hashed with the query string using md5 and the resulting md5 hash added to the end of the query string as md parameter.

acv, 2015-05-11:
If we do that, it'll be HMAC-SHA2-256. No sense using broken 1990s technolofy in 2015.

acv, 2015-05-11:
Is tampering an actual problem?

gareth41, 2015-05-11:
Tampering with the auth query string I would think shouldn't be a problem, but users could quite easily change parameters in the redirect query string. I currently use coovachilli and wrote a login script in php for a hotspot to offer free wifi between certain times with a data limit per mac per 24hr. The login/landing page uses the device mac in query string to do some checks.

acv, 2015-05-11:
They can tamper with the MAC but when the gateway asks the auth server to validate the token, the validation will fail if the MAC is different and the gateway uses ARP to get the client's MAC.

gareth41, 2015-05-12:
of couse, it makes sense. it could work by login script generating token and saving to db with the mac address from login/redirect query string. upon login the token is sent to auth server by wifidog and auth server checks the mac already in db matches the arp mac sent in the auth request. if they dont match user is denied auth.

acv, 2015-05-12:
That was the idea and the first implementation, way back in 2004.

That said, I like the idea generally and we now have optional SSL support so if CyaSSL exports crypto primitives like HMAC, it could well be implemented! I'm going to leave the issue open with no milestone. A lot of us are swamped at the moment with IRL stuff so progress has slowed compared to March but this is a cool feature so I expect it gets implemented soon.

Patches welcome. Prospective patches should:

1. Use cyassl for crypto code since it's already a dependency.
2. Use HMAC construction. Much better then straight hash + secret, it's secure even when hash function no longer provides collision resistance (within limits, HMAC with MD5 is limping, with MD4 is breakable, while HMAC using SHA-1 is safe)
3. Use SHA2-256 for hash. No sense in building anything new broken MD5 and weak SHA-1.
4. Figure out a way to organize the payload so it's deterministically verifiable, even if we add future arguments to the payload (basically a deterministic ordering.)

gareth41, 2015-05-12:
Thanks, looking forward to it.  Unfortunately i'm not very clued up on c++ but im learning more, i've always been a php/mysql man.