wifidog/wifidog-gateway #209: mac authentication

gareth41, 2015-05-13:
There could also be an option to do mac authentication.  Upon a client joining the network and requesting DHCP, wifidog will immediately send an auth request with the mac, the auth server should have the mac tied with a user account.  The device is authenticated and user does not see the captive portal. There could also be a config parameter to try mac auth x number of seconds for each connected client that hasn't yet authenticated themselves.

acv, 2015-05-13:
There's already a MAC authentication (whitelist) feature.

databeille, 2015-05-13:
... and I would not say MAC authentication is authentication :/

gareth41, 2015-05-13:
I understand there is already a mac white list, however this takes mac addresses from the config file.  I'm thinking along the lines of being able to allow wifi users to white list their own macs after logging in with their account and not just white listing on the router/gateway they are connected to but all routers/wifidog gateways using the same auth server.  This is far easier if the auth server db can hold a list of mac addresses attached to user accounts, and wifi dog performs an automatic auth request upon device joining the network/obtaining dhcp.  coovachilli supports this and it works very well, however because coova-chilli doesn't use tokens it just sends the mac as the username to its configured radius servers and the radius server matches the mac with the correct username.

acv, 2015-05-13:
Not gonna happen. MAC are entirely too dicey. Please nothing prevents the auth server from implementing a policy like that.

closed 2015-05-13 by acv