wifidog/wifidog-gateway #210: Configuration from server

t123yh, 2015-05-13:
I've seen #209 and I am thinking about the possibility of the WiFiDog to get some configuration such as the white-list from the auth server. It should be much more convenient if somebody can implement that.

databeille, 2015-05-13:
I am (slowly) working on such a thing, using authpuppy (http://www.authpuppy.org/) and apDashboardPlugin (https://code.launchpad.net/~alliancecsf-dev/authpuppy/apDashboardPlugin).

See my fork for the gateway scripts : https://github.com/databeille/apDashboardPlugin_scripts
The plugin itself is rather uncomplete/buggy, but would probably be fixed easily.

The drawback : it is openWRT only.

You're welcome to give an hand :)

t123yh, 2015-05-13:
I have no knowledge at PHP at all but I've been writing an auth server using ASP.NET MVC. I'm busy with other things these days. A week later I'll be able to continue writing this authentication server. I think at that time I can offer some help.

gareth41, 2015-05-13:
I have written a php script to output a bash script via http that will execute on a router running openwrt.  The php script will query a mysql db for a list of awaiting bash commands relating to the router id.  The router uses cron to ping this php script every 5min or so using wget, upon php script sending bash script to router, db is then updated to indicate commands were executed.  If anyone wants a copy please let me know.

wendal, 2015-05-17:
maybe my pr will help https://github.com/wifidog/wifidog-gateway/pull/212

iffifan, 2017-03-17:
any update on this?

wendal, 2017-03-17:
seem the project is dead

KazukiShimada, 2017-03-17:
apfree_wifidog, a fork of wifidog gateway, is being impremented such a feature on going.
https://github.com/liudf0716/apfree_wifidog
It seems mqtt is used.
https://github.com/liudf0716/apfree_wifidog/blob/master/apfree_wifidog_mqtt_protocol

mhaas, 2017-03-18:
@wendal project is in maintenance mode, rather than dead :) I personally do not have capacity right now beyond code-reviewing merge requests and merge the ones that are ready.