wifidog/wifidog-gateway #213: Add a new config section to switch on new protocol

t123yh, 2015-05-19:
Many actions in the WiFiDog Protocol is lacking in features. For example, the PortalScriptPathFragment does not provide a token of user for the server. That makes the redirection to the url the user is trying to fetch before signing in hard. But if we modify the protocol directly there may be compatibility problems. So can we just add a new section in the configuration file such as "NewProtocol yes" to enable new features added to WiFiDog?

mhaas, 2015-05-20:
Adding a new query parameters probably should not break stuff.

Couldn't the portal website simply set a cookie with the redirect url? I am a bit hazy on the current implementation right now.

acv, 2015-10-01:
I'm not really seeing what is to do here.

closed 2015-10-01 by acv