wifidog/wifidog-gateway #214: make package file

turanalmammadov, 2015-05-25:
Hello.
I have just installed OpenWrt to my router.
I want to install the latest release of wifidog (1.2.1).
I am downloading file but ./configure and make is not working.
I think I need to create ipk file for installing the latest version.
Is there someone going to explain step by step installing the latest version from github to OpenWrt Barrier Breaker 14.07

t123yh, 2015-05-25:
I can, but later.

turanmete16, 2015-05-25:
will wait :+1:

turanmete16, 2015-05-25:
Dear t123yh :)
Can u please help us on this problem :)

turanmete16, 2015-05-25:
I am using ubuntu downloaded openwrt and build it like in http://wiki.openwrt.org/doc/howto/buildroot.exigence
made "make menuconfig" as a Openwrt SDK but then I dont know what to do to compile the latest wifidog and take this ipk file to my router.

t123yh, 2015-05-26:
Oh, have a look at this: https://github.com/JiapengLi/OpenWrt-gmediarender
This is a patch that compiles gmediarender on openwrt.
Try to use this on WiFiDog. Use the makefile from here:
https://github.com/wifidog/packages/tree/wifidog-experimental/net/wifidog

mhaas, 2015-05-26:
Use the Wifidog package file in OpenWrt chaos calmer. We recently brought Wifidog back to OpenWrt :)

turanmete16, 2015-05-26:
I downloaded package 
make package/wifidog/download
make package/example/prepeare

I am getting error while compiling

root@ubuntu:~/openwrt# make package/wifidog/compile -j1 V=s
make[1]: Entering directory `/root/openwrt'
make[2]: Entering directory `/root/openwrt/package/libs/toolchain'
mkdir -p /root/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/toolchain
touch /root/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/toolchain/.prepared_4357c9ca2fe557e05cd5f32c2016e8e6
rm -f /root/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.toolchain_installed
(cd /root/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/toolchain/./; if [ -x ./configure ]; then find /root/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/toolchain/ -name config.guess | xargs -r chmod u+w; find /root/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/toolchain/ -name config.guess | xargs -r -n1 cp --remove-destination /root/openwrt/scripts/config.guess; find /root/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/toolchain/ -name config.sub | xargs -r chmod u+w; find /root/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/toolchain/ -name config.sub | xargs -r -n1 cp --remove-destination /root/openwrt/scripts/config.sub; AR="mips-openwrt-linux-uclibc-gcc-ar" AS="mips-openwrt-linux-uclibc-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float" LD=mips-openwrt-linux-uclibc-ld NM="mips-openwrt-linux-uclibc-gcc-nm" CC="mips-openwrt-linux-uclibc-gcc" GCC="mips-openwrt-linux-uclibc-gcc" CXX="mips-openwrt-linux-uclibc-g++" RANLIB="mips-openwrt-linux-uclibc-gcc-ranlib" STRIP=mips-openwrt-linux-uclibc-strip OBJCOPY=mips-openwrt-linux-uclibc-objcopy OBJDUMP=mips-openwrt-linux-uclibc-objdump SIZE=mips-openwrt-linux-uclibc-size CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 " CPPFLAGS="-I/root/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include -I/root/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/include -I/root/openwrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/root/openwrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/include " LDFLAGS="-L/root/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/lib -L/root/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/lib -L/root/openwrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/lib -L/root/openwrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/lib "   ./configure --target=mips-openwrt-linux --host=mips-openwrt-linux --build=x86_64-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls   ; fi; )
rm -f /root/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/toolchain/.configured_*
touch /root/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/toolchain/.configured_yyyyyyyy
cp -fpR /root/openwrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/lib/libuClibc-*.so /root/openwrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/lib/libcrypt-*.so /root/openwrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/lib/libm-*.so /root/openwrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/lib/libpthread-*.so /root/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/toolchain/
cp: cannot stat '/root/openwrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/lib/libuClibc-*.so': No such file or directory
cp: cannot stat '/root/openwrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/lib/libcrypt-*.so': No such file or directory
cp: cannot stat '/root/openwrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/lib/libm-*.so': No such file or directory
cp: cannot stat '/root/openwrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/lib/libpthread-*.so': No such file or directory
make[2]: *** [/root/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/toolchain/.built] Error 1
make[2]: Leaving directory `/root/openwrt/package/libs/toolchain'
make[1]: *** [package/libs/toolchain/compile] Error 2
make[1]: Leaving directory `/root/openwrt'
make: *** [package/wifidog/compile] Error 2
root@cgi-proxy:~/openwrt#

turanmete16, 2015-05-27:
still getting same error

mhaas, 2015-05-28:
That looks like a problem with your buildroot. Can you build other packages?

Also, building as root is a bad idea.

databeille, 2015-06-01:
@turanmete16 I stopped cross compiling with Ubuntu, because of many errors I did not want to solve. I use Fedora 20 x86_64, and it works fine with http://wiki.openwrt.org/doc/howto/obtain.firmware.sdk

It seems your toolchain is missing.

acv, 2015-10-01:
Closing this support issue with OpenWrt buildroot.

closed 2015-10-01 by acv