wifidog/wifidog-gateway #215: An Issue on running wifidog-20130917 with OpenWRT-1407- can not portal

hbl0307106015, 2015-05-28:
Hi, all,
I am using wifidog-20130917 openwrt-1407. 
The captive portal can not be opened whatever wireless client connect to my ap.

There is a br-lan interface with eth0.1(LAN) and wlan0, 
I just want to test wifidog in my local network wiht LAN. below is my configuration file:

GatewayID default
ExternalInterface eth0.1
GatewayInterface br-lan
HtmlMessageFile /etc/wifidog-msg.html
AuthServer {
	Hostname 192.168.1.100
	SSLavailable no
    Path /
}

ClientTimeout 7200

FirewallRuleSet global {

}
FirewallRuleSet validating-users {
FirewallRule allow to 0.0.0.0/0
}
FirewallRuleSet known-users {
FirewallRule allow to 0.0.0.0/0
}
FirewallRuleSet unknown-users {
FirewallRule allow udp port 53
FirewallRule allow tcp port 53
FirewallRule allow udp port 67
FirewallRule allow tcp port 67
}
FirewallRuleSet locked-users {
FirewallRule block to 0.0.0.0/0
}

And my log is :
[6][Tue May 26 12:37:52 2015]9873 Received connection from 192.168.1.172, spawning worker thread
[7][Tue May 26 12:37:52 2015]9873 No valid request received from 192.168.1.172
[7][Tue May 26 12:37:52 2015]9873 Closing connection with 192.168.1.172
[6][Tue May 26 12:37:52 2015]9873 Received connection from 192.168.1.172, spawning worker thread
[7][Tue May 26 12:37:52 2015]9873 No valid request received from 192.168.1.172
[7][Tue May 26 12:37:52 2015]9873 Closing connection with 192.168.1.172
[6][Tue May 26 12:37:57 2015]9873 Received connection from 192.168.1.172, spawning worker thread
[7][Tue May 26 12:37:57 2015]9873 No valid request received from 192.168.1.172
[7][Tue May 26 12:37:57 2015]9873 Closing connection with 192.168.1.172
[6][Tue May 26 12:37:57 2015]9873 Received connection from 192.168.1.172, spawning worker thread
[7][Tue May 26 12:37:57 2015]9873 Processing request from 192.168.1.172
[7][Tue May 26 12:37:57 2015]9873 Calling httpdProcessRequest() for 192.168.1.172
[6][Tue May 26 12:37:57 2015]9873 Got client MAC address for ip 192.168.1.172: 9c:99:a0:40:df:4e
[6][Tue May 26 12:37:57 2015]9873 Captured 192.168.1.172 requesting [http%3A//dns.weixin.qq.com/cgi-bin/micromsg-bin/newgetdns%3Fuin%3D0%26clientversion%3D637599817%26scene%3D0%26net%3D1%26md5%3Dbbfad2a27141b1fe1a2f6d252e55763a%26devicetype%3Dandroid-19%26lan%3Den%26sigver%3D1] and re-directing them to login page
[7][Tue May 26 12:37:57 2015]9873 Redirecting client browser to http://192.168.1.100:80/login/?gw_address=192.168.1.1&gw_port=2060&gw_id=143BCEEC320&mac=9c:99:a0:40:df:4e&url=http%3A//dns.weixin.qq.com/cgi-bin/micromsg-bin/newgetdns%3Fuin%3D0%26clientversion%3D637599817%26scene%3D0%26net%3D1%26md5%3Dbbfad2a27141b1fe1a2f6d252e55763a%26devicetype%3Dandroid-19%26lan%3Den%26sigver%3D1
[7][Tue May 26 12:37:57 2015]9873 Returned from httpdProcessRequest() for 192.168.1.172
[7][Tue May 26 12:37:57 2015]9873 Closing connection with 192.168.1.172
[6][Tue May 26 12:37:57 2015]9873 Received connection from 192.168.1.172, spawning worker thread
[7][Tue May 26 12:37:57 2015]9873 Processing request from 192.168.1.172
[7][Tue May 26 12:37:57 2015]9873 Calling httpdProcessRequest() for 192.168.1.172
[6][Tue May 26 12:37:57 2015]9873 Got client MAC address for ip 192.168.1.172: 9c:99:a0:40:df:4e
[6][Tue May 26 12:37:57 2015]9873 Captured 192.168.1.172 requesting [http%3A//szshort.weixin.qq.comhttp%3A/szshort.weixin.qq.com/cgi-bin/micromsg-bin/newauth] and re-directing them to login page
[7][Tue May 26 12:37:57 2015]9873 Redirecting client browser to http://192.168.1.100:80/login/?gw_address=192.168.1.1&gw_port=2060&gw_id=143BCEEC320&mac=9c:99:a0:40:df:4e&url=http%3A//szshort.weixin.qq.comhttp%3A/szshort.weixin.qq.com/cgi-bin/micromsg-bin/newauth
[7][Tue May 26 12:37:57 2015]9873 Returned from httpdProcessRequest() for 192.168.1.172
[7][Tue May 26 12:37:57 2015]9873 Closing connection with 192.168.1.172
[6][Tue May 26 12:38:00 2015]9873 Received connection from 192.168.1.172, spawning worker thread
[7][Tue May 26 12:38:00 2015]9873 Processing request from 192.168.1.172
[7][Tue May 26 12:38:00 2015]9873 Calling httpdProcessRequest() for 192.168.1.172
[6][Tue May 26 12:38:00 2015]9873 Got client MAC address for ip 192.168.1.172: 9c:99:a0:40:df:4e
[6][Tue May 26 12:38:00 2015]9873 Captured 192.168.1.172 requesting [http%3A//111.13.142.10%3A80/mibind/http-bind%3Ft%3D197] and re-directing them to login page
[7][Tue May 26 12:38:00 2015]9873 Redirecting client browser to http://192.168.1.100:80/login/?gw_address=192.168.1.1&gw_port=2060&gw_id=143BCEEC320&mac=9c:99:a0:40:df:4e&url=http%3A//111.13.142.10%3A80/mibind/http-bind%3Ft%3D197
[7][Tue May 26 12:38:00 2015]9873 Returned from httpdProcessRequest() for 192.168.1.172
[7][Tue May 26 12:38:00 2015]9873 Closing connection with 192.168.1.172
[6][Tue May 26 12:38:02 2015]9873 Received connection from 192.168.1.172, spawning worker thread
[7][Tue May 26 12:38:02 2015]9873 No valid request received from 192.168.1.172
[7][Tue May 26 12:38:02 2015]9873 Closing connection with 192.168.1.172
[6][Tue May 26 12:38:10 2015]9873 Received connection from 192.168.1.172, spawning worker thread
[7][Tue May 26 12:38:10 2015]9873 Processing request from 192.168.1.172
[7][Tue May 26 12:38:10 2015]9873 Calling httpdProcessRequest() for 192.168.1.172
[6][Tue May 26 12:38:10 2015]9873 Got client MAC address for ip 192.168.1.172: 9c:99:a0:40:df:4e
[6][Tue May 26 12:38:10 2015]9873 Captured 192.168.1.172 requesting [http%3A//111.13.142.10%3A80/mibind/http-bind%3Ft%3D634] and re-directing them to login page
[7][Tue May 26 12:38:10 2015]9873 Redirecting client browser to http://192.168.1.100:80/login/?gw_address=192.168.1.1&gw_port=2060&gw_id=143BCEEC320&mac=9c:99:a0:40:df:4e&url=http%3A//111.13.142.10%3A80/mibind/http-bind%3Ft%3D634
[7][Tue May 26 12:38:10 2015]9873 Returned from httpdProcessRequest() for 192.168.1.172
[7][Tue May 26 12:38:10 2015]9873 Closing connection with 192.168.1.172
[6][Tue May 26 12:38:12 2015]9873 Received connection from 192.168.1.172, spawning worker thread

hbl0307106015, 2015-05-28:
Maybe I should use the openwrt-trunk with wifidog-1.2.1 ?

hbl0307106015, 2015-05-28:
I tried on openwrt trunk and wifidog 1.2.1, same problem here.

turanmete16, 2015-05-28:
Can you send wifidog 1.2.x ipk file.

hbl0307106015, 2015-05-28:
You mean upload attachment file ?  github can not upload that file

turanmete16, 2015-05-28:
mail

hbl0307106015, 2015-05-28:
Ok, please check you mailbox, I sent it.

hbl0307106015, 2015-05-28:
I thought I was succeed to make it, I was wrong. Neither the authpuppy nor auth-server can make the captive portal work fine.  My android phones can not be redirected to the auth web.

Both the auth-server and authpuppy can reply pong packet to the router, however, the wireless client can not open the portal page (login).

Any hints willed appreciated!!! really thanks.

turanmete16, 2015-05-28:
I was making the same I also got the error.
I think we must compile the 1.2.0.
Because it works fine.

hbl0307106015, 2015-05-28:
By the way, what your hardware for your auth server (or authpuppy), mine is ubuntu - 12.04 64 bit.

turanmete16, 2015-05-28:
I am using Centos for auth script.
But I wrote my own.

hbl0307106015, 2015-05-28:
All right, I just want to have a look at the original solution. Getting stuck.!!!!  :(   :(   T.T

turanmete16, 2015-05-28:
what is your router model ?

hbl0307106015, 2015-05-28:
My router is MT7620a,  br-lan (bridge)  with  eht0.1  LAN interfaces.

hbl0307106015, 2015-05-28:
I tried on openwrt-1407 with wifidog-20130917, failed again.  The same issue shown above arises.  :(  :(

turanmete16, 2015-05-28:
It means it can be because of your router.
Are you sure that you properly installed ipk file to router ?

hbl0307106015, 2015-05-28:
Yes, I am sure, I tried it on another router which can buy all over the market. Lenovo Y1

turanmete16, 2015-05-28:
Just wait I will send you proper answer

hbl0307106015, 2015-05-28:
Do not forget to send my wifidog-1.2.0 which works fine.

turanmete16, 2015-05-28:
ok :)

databeille, 2015-06-01:
Hi guys, I use authpuppy and wifidog 1.2.1 trunk on openwrt 14.07 to run captive portals.
Everything works fine.

Please check your server for authentication is working properly and use a minimal wifidog config file (remove all that is not mandatory, like ExternalInterface).

Wifidog 1.2.0 is broken : https://github.com/wifidog/wifidog-gateway/releases/tag/1.2.0

hbl0307106015, 2015-06-02:
Now I know the reason why wifidog can not show portal on phones, because the wifidog-gateway can not access to the internet.

databeille, 2015-06-02:
It is a good point ! Try to force a local trusted server in your config file.

See https://github.com/wifidog/wifidog-gateway/pull/203

acv, 2015-10-01:
This seems to have been a support issue, since it seems dealt with I'm closing.

closed 2015-10-01 by acv

DineshSudham, 2016-04-18:
Hi,
I am working on authpuppy+wifidog to enable vaptive portal on openwrt WLAN router, Please anyone suggest me how to do, I installed the authpuppy on my ubuntu machine, wifidog is running on router, wifidog able communicate with the authpuppy server.But i am not getting the splash page,Can anyone suggest how to achieve this.Thanks in advance.

databeille, 2016-04-18:
@DineshSudham check your logs (apache or nginx or http server your are using) and set them to be verbose.
Are you able to connect as an admin ? Did you clean symfony cache ?
It does not seem to be wifidog related.

hbl0307106015, 2016-04-18:
@databeille  I solved this problem, it is about dns-server setting, in china, google.com is not accessible thus the dns initialization is not succeed, I added another one.

najwa2, 2018-07-20:
I work on a captive portal wifidog on openwrt using tplink tl-mr3020.
the installation of wifidog went well but when debug it, it seems like it couldn't create a web server on the ip adress of the openwrt and it shows the error below:
could not create web server : addess already in use
I only have br-lan interface.
I linked the openwrt router that has as ip address as 192.168.1.4 to wifi router that has 192.168.1.1.
I couldn't detect the error
[image]

lgaitan, 2018-07-20:
@najwa2 

It looks like you have another instance of wifidog running.

Try running:
```shell
netstat -tulpn | grep 2060
```
to see the process id and then kill it

najwa2, 2018-07-24:
Hi @lgaitan  
Thank you so much 
Actually it works, the problem was other instances running  and the server says pong but when other devices are connected to the wifi the authentication page doen't show up and it's directly connected to wifi, do you have any idea why ?