wifidog/wifidog-gateway #216: //usr/lib/opkg/info/wifidog.postinst: line 4: default_postinst: not found

turanmete16, 2015-05-28:
while installing wifidog 1.2.1 on Barrier Breaker 14.07 I am getting this error 

Installing wifidog (1.2.1-1) to root...
Configuring wifidog.
//usr/lib/opkg/info/wifidog.postinst: line 4: default_postinst: not found
Collected errors:
 * pkg_run_script: package "wifidog" postinst script returned status 127.
 * opkg_configure: wifidog.postinst returned 127.

mhaas, 2015-05-28:
Where did you get the package?

turanmete16, 2015-05-28:
from github. 1.2.1

package/wifidog/download

mhaas, 2015-05-28:
How did you build it?

turanmete16, 2015-05-28:
firstly downloaded and built openwrt from github (wiki.openwrt.org/doc/howto/buildroot.exigence)
then made menuconfig (choosed Openwrt SDK and Packages/wifidog)
then make
then I took ipk file from /bin.

mhaas, 2015-05-28:
Is it possible that you built a package for OpenWrt Chaos Calmer and tried to install it on OpenWrt Barrier Breaker? As far as I know, Wifidog does not supply its own postinst script. As such, the failure likely originates somewhere else -- or the packaging script is broken.

turanmete16, 2015-05-28:
Yes its possible.
Also I want to add that even if I am facing this error wifidog is working properly.
Also is the any proper solution for social network logins ?

strong46066999, 2015-12-17:
I have seen wifidog is in openwrt 15.01 feeds. just ./script/feeds install wifidog          ok, is it easyer ?