wifidog/wifidog-gateway #218: wdctl -s /tmp/wdtest.sock restart not work

dhpark21, 2015-06-08:
wdctl -s /tmp/wdtest.sock restart not work: (did not redirect)

but, /etc/rc.d/init.d/wifidog stop 
and /etc/rc.d/init.d/wifidog restart work

so did not work "Re-start the running wifidog (without disconnecting active users!)"
Thanks

databeille, 2015-06-09:
openWRT >= 14.07 uses procd to manage daemons.
In fact, wdctl does probably not comply with the way procd handles processes.
It has to be fixed or removed if we can do the same with procd.

imnobodyet, 2015-06-12:
The same issue here, compiled wifidog release 1.2.1 on ubuntu 14.04 64bit.

Using "wdctl restart", the wifidog prints no error, but the web socket binding is not there with "netstat". 

I found out the web socket fd was always 0 after a wdctl restart, so for a temporary solution: 

add following codes in libhttpd/api.c : 265
       // no 0,1,2 for fd
	while(sock < 3){
		sock = socket(AF_INET, SOCK_STREAM, 0);
		if (sock < 0) {
			free(new);
			return (NULL);
		}
	}

gareth41, 2015-08-24:
I'm getting the same problem on Kamikaze 8.09.1 running on a linksys WRT654GL, wifidog appears to download client list and no errors are reported.  Webserver also appears to start again after forking, but in reality webserver is not running and netstat -l confirms no webserver is listening on wifidog gateway address:port, hence redirection for clients doesn't work.

I tried changing the above code which didn't seem to fix the problem.