wifidog/wifidog-gateway #219: help creating ipk for openwrt

dirkam, 2015-06-09:
Hi,

I'd like to modify something in the source code and create an ipk package that I can install on OpenWRT (BB). I've tried it, but I've never done this before and I couldn't manage to find a solution.

Would any of you guys please guide me how could I do it?

Thanks.

KazukiShimada, 2015-06-10:
It's easy to build wifidog ipk on openwrt buildroot.
Get it
https://dev.openwrt.org/wiki/GetSource
build it
http://wiki.openwrt.org/doc/howto/build

dirkam, 2015-06-10:
Would you please elaborate on this?
I've downloaded 14.07 branch and followed the instructions on the links.
I guess I have the buildroot but how should I proceed?
I tried to download the wifidog 1.2.1 source, modify it based on my needs and copy the complete folder to packages/network/services. But I guess it's not that simple.
I tried to run ./autogen.sh and make in the wifidog1.2.1 folder but I can't seem to find the ipk.

I appreciate any help you can provide. Thanks

mhaas, 2015-06-11:
You may find my build scripts helpful: https://github.com/mhaas/wifidog-openwrt-builder

dirkam, 2015-06-11:
@mhaas Thanks for the script. I ran it, SDK and wifidog package was downloaded but I can't find SDK/bin folder.
When I run ./build.sh I get this error at the end:

````
make[3]: Entering directory `/home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/build_dir/target-mips_34kc_uClibc-0.9.33.2/wifidog-1434010717'
make[3]: *** No targets specified and no makefile found.  Stop.
make[3]: Leaving directory `/home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/build_dir/target-mips_34kc_uClibc-0.9.33.2/wifidog-1434010717'
make[2]: *** [/home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/build_dir/target-mips_34kc_uClibc-0.9.33.2/wifidog-1434010717/.built] Error 2
make[2]: Leaving directory `/home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/feeds/wifidog_gateway/net/wifidog'
make[1]: *** [package/feeds/wifidog_gateway/wifidog/install] Error 2
make[1]: Leaving directory `/home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2'
make: *** [package/wifidog/install] Error 2
#
# configuration written to .config
#
 make[1] package/index
````

mhaas, 2015-06-11:
Can you post the full log?

dirkam, 2015-06-11:
````
make[2]: Entering directory `/home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/feeds/wifidog_gateway/net/wifidog'
mkdir -p /home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/dl
echo "Checking out files from the git repository..."; mkdir -p /home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/tmp/dl && cd /home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/tmp/dl && rm -rf wifidog-1434010717 && [ \! -d wifidog-1434010717 ] && git clone git://github.com/wifidog/wifidog-gateway.git wifidog-1434010717 --recursive && (cd wifidog-1434010717 && git checkout b63ebe4e8294b01d8071db58872a84a0386a21cb && git submodule update) && echo "Packing checkout..." && rm -rf wifidog-1434010717/.git &&   /bin/gtar czf /home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/tmp/dl/wifidog-1434010717-b63ebe4e8294b01d8071db58872a84a0386a21cb.tar.gz wifidog-1434010717 && mv /home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/tmp/dl/wifidog-1434010717-b63ebe4e8294b01d8071db58872a84a0386a21cb.tar.gz /home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/dl/ && rm -rf wifidog-1434010717; 
Checking out files from the git repository...
Cloning into 'wifidog-1434010717'...
Note: checking out 'b63ebe4e8294b01d8071db58872a84a0386a21cb'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b new_branch_name

HEAD is now at b63ebe4... Merge pull request #211 from t123yh/DeltaTraffic
Packing checkout...
. /home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/include/shell.sh; gzip -dc /home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/dl/wifidog-1434010717-b63ebe4e8294b01d8071db58872a84a0386a21cb.tar.gz | /bin/gtar -C /home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/build_dir/target-mips_34kc_uClibc-0.9.33.2/wifidog-1434010717/.. -xf - 
touch /home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/build_dir/target-mips_34kc_uClibc-0.9.33.2/wifidog-1434010717/.prepared_f49da34d8d9f4aed99b26748521e2d5b
(cd /home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/build_dir/target-mips_34kc_uClibc-0.9.33.2/wifidog-1434010717; rm -f aclocal.m4; if [ -f ./configure.ac ] || [ -f ./configure.in ]; then [ -d ./autom4te.cache ] && rm -rf autom4te.cache; [ -e ./config.rpath ] || ln -s /home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/scripts/config.rpath ./config.rpath; touch NEWS AUTHORS COPYING ABOUT-NLS ChangeLog; AUTOM4TE=/home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/staging_dir/host/bin/autom4te AUTOCONF=/home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/staging_dir/host/bin/autoconf AUTOMAKE=/home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/staging_dir/host/bin/automake ACLOCAL=/home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/staging_dir/host/bin/aclocal AUTOHEADER=/home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/staging_dir/host/bin/autoheader LIBTOOLIZE=/home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/staging_dir/host/bin/libtoolize LIBTOOL=/home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/staging_dir/host/bin/libtool M4=/home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/staging_dir/host/bin/m4 AUTOPOINT=true /home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/staging_dir/host/bin/autoreconf -v -f -i -s -B /home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/staging_dir/host/share/aclocal -I /home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/staging_dir/target-mips_34kc_uClibc-0.9.33.2/host/share/aclocal -I /home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/share/aclocal -I m4 -I . . || true; fi; );
autoreconf: Entering directory `.'
autoreconf: configure.in: not using Gettext
autoreconf: running: /home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/staging_dir/host/bin/aclocal -I /home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linu [... cut ...]
make[3]: Leaving directory `/home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/build_dir/target-mips_34kc_uClibc-0.9.33.2/wifidog-1434010717'
make[2]: *** [/home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/build_dir/target-mips_34kc_uClibc-0.9.33.2/wifidog-1434010717/.built] Error 2
make[2]: Leaving directory `/home/user/barrier_breaker/14.07/ar71xx/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/feeds/wifidog_gateway/net/wifidog'
````

dirkam, 2015-06-12:
Can someone please upload 1.2.1 ipk somewhere or send it to me by email?

I still would like to make some changes to the code and create an ipk from it but I also want to test something with 1.2.1.

Thanks

dirkam, 2015-06-13:
I was able to run your build script, libtool and ccache was missing. Thanks for the script.
Next step is to figure out how could I use it to build my own source code.

gareth41, 2015-06-13:
Try adding the following to your Makefile and it should work, it compiled for me without any problems.
PKG_FIXUP:=autoreconf

I have the ipk file for ar71xx since this is the hardware i'm using. I have uploaded it here http://www.gareth.net.nz/openwrt/wifidog_1.2.1-2_ar71xx.ipk

I can compile for others like broadcom (linksys) if required, but ideally you should compile yourself as then you can do a lot more with customizing the router firmware etc... with config files already set to go, no need to have to configure each individual router from start if you're deploying a whole lot.

mhaas, 2015-06-13:
I personally commit the code to my github and then build that branch in the script. You can also just run 'make package/wifidog/build' in the buildroot (which is located somewhere below the script dir) once you adjust the Rev in the Makefile.

kuzjnvn, 2015-07-20:
i have openwrt 12.09 , i dont know how to have wifidog 1.2.1 ipk to install , my wifidog is old :( can you help me ?

mhaas, 2015-08-16:
What architecture do you have?

raxis, 2015-09-29:
Hi mhaas, 
i need ipk for my openwrt 12.09 tl mr 740 , when i install opkg wifidog, it seem old installer. so i need to install latest wifi dog, 

i follow your script in : https://github.com/mhaas/wifidog-openwrt-builder but i have no idea what should i do next, i got OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2.tar.bz2 after build.

missionrulz, 2016-02-02:
Hi @mhaas ,

I tried to use your build script with attitude adjustment. Everything downloads correctly, however it fails on the following. Any ideas?

```
make[3]: Entering directory `/home/missionrulz/wifidog/test/attitude_adjustment/12.09/brcm47xx/OpenWrt-SDK-brcm47xx-for-linux-i486-gcc-4.6-linaro_uClibc-0.9.33.2/scripts/config'
make[3]: *** No rule to make target `mconf.o', needed by `mconf'.  Stop.
make[3]: Leaving directory `/home/missionrulz/wifidog/test/attitude_adjustment/12.09/brcm47xx/OpenWrt-SDK-brcm47xx-for-linux-i486-gcc-4.6-linaro_uClibc-0.9.33.2/scripts/config'
make[2]: *** [scripts/config/mconf] Error 2
make[1]: *** [.config] Error 2
make: *** [package/wifidog/install] Error 2
```

mhaas, 2016-02-04:
@missionrulz No clue - that's probably an issue between the 12.09 SDK and your linux install :(

missionrulz, 2016-02-05:
@mhaas, no worries. It turned out to be some misconfiguration within the SDK for attitude adjustment (yet to get it working). However, your script worked flawlessly for both barrier breaker & chaos calmer.

demslob, 2016-02-13:
@mhaas 
Sorry me for my poor knowledge, but i don't know such things - i should run this script on my openWRT router or on my Linux PC? Should i have another files besides this one to do this?
i have BB (14.07, r42625) router
Thank you very much!

demslob, 2016-02-14:
@gareth41 Hey, you said you can compile for others, can you please compile for ramips mt7620a?

gareth41, 2016-02-14:
Which version of openwrt did you want it compiled for?

demslob, 2016-02-14:
@gareth41 as you want 14.07 or 15.05

demslob, 2016-02-14:
@gareth41 thank you so much for helping me!

demslob, 2016-02-14:
@gareth41 can you please send it via e-mail?

demslob, 2016-02-14:
@gareth41 can you please make for BB 14.07 version

gareth41, 2016-02-15:
http://www.gareth.net.nz/openwrt/barrier_breaker/ramips/packages/oldpackages/wifidog_20160130-1b835bada57d948d70cd269fe0cf9d16f27fe8de_ramips_24kec.ipk
http://www.gareth.net.nz/openwrt/chaos_calmer/ramips/packages/packages/wifidog_20160130-1b835bada57d948d70cd269fe0cf9d16f27fe8de_ramips_24kec.ipk
Both compiled from the last commit

Here are the full openwrt images with wifidog pre installed.  I don't have any ramips devices so i'm unable to test these.
http://www.gareth.net.nz/openwrt/barrier_breaker/ramips/
http://www.gareth.net.nz/openwrt/chaos_calmer/ramips/

demslob, 2016-02-15:
@gareth41 Than you very much! I will test it right now and tell the result.

demslob, 2016-02-15:
@gareth41 everything works fine, thanks again!

deivisonrpg, 2016-02-15:
@gareth41 you can help me on this plz? #243 Tks in man