wifidog/wifidog-gateway #220: Time to update the offical document?

t123yh, 2015-06-14:
I am writing an ASP.NET-MVC-based authentication server, and I'm thinking about when to log off a token from the database. No one told me that the AuthScriptPathFragment has a logout stage until I used the Wireshark to capture the packet for debug use just now. I think it's time to update the WiFiDog offical document (http://dev.wifidog.org/wiki/doc/developer/WiFiDogProtocol_V1), or else there will be more developers like me stuck on some important problems.

acv, 2015-06-19:
I'm not sure who has access to the old ISF/wifidog wiki anymore. Maybe @benoitg ?

We should probably just make a new document with the current proto part of the repo.

benoitg, 2015-08-29:
@acv Everything should be moved here.  Being stuck on ISF's unmaintained infrastructure isn't going to help the project...