wifidog/wifidog-gateway #228: Interested in the Prerouting portion on Wifidog

iDVB, 2015-08-12:
I'm interested in setting things up so that when people connect to my Wifi AP (RaspberryPi) they automatically get a browser popup that directs them to my AP's auth page. Similar to what you see for coffee shops etc. 

Is there a way to set this up using wifidog.... or maybe its possible natively in unix?

t123yh, 2015-08-12:
Just use default WiFiDog's settings, then the authorization page will appear automatically.
The mobile phone sends a request to its Internet-connection-checking-server, and if it gets a redirect request, then it will show the login page (destination of the redirect).

iDVB, 2015-08-13:
Thanks @t123yh is there a way to get just that portion working? I already have a nodejs server that I want to serve up the actual page that things get redirected to.

mhaas, 2015-08-16:
Is your Raspberry Pi different from your AP, i.e. do you have two physical devices there?
What does your nodejs server do?

To phrase my question differently: do you already have a fully working captive portal and just want people connecting to rasppi to use that?

iDVB, 2015-08-24:
Nope the Rpi and AP are one and the same thing. And yes, I already have an nodejs web application and just want people to be default driven to it as well as locked to it, all other urls blocked.

iDVB, 2015-09-08:
Essentially, I just want it to ALWAYS redirect to my server. I do NOT want people to access any other site then the local one I default them to.

mhaas, 2015-09-12:
That is easily done with a single iptables rule: http://unix.stackexchange.com/questions/203791/how-to-redirect-all-http-request-to-a-local-web-server

iDVB, 2015-09-12:
Thanks! I'll give this a go

closed 2015-09-12 by iDVB