wifidog/wifidog-gateway #230: wifidog-gateway-master 1.2.1 wdctl restart error

gigibox, 2015-08-26:
这个版本wifidog启动以后，使用wdctl restart命令重启的话，webserver->serverSock会被关闭，无法重定向client的http请求，不知道其他人有没有遇到这个问题

gigibox, 2015-08-26:
############################################################################
The same issue here, compiled wifidog release 1.2.1 on ubuntu 14.04 64bit.

Using "wdctl restart", the wifidog prints no error, but the web socket is not there with "netstat".

I found out the web socket fd was always 0 after a wdctl restart, so for a temporary solution:

add following codes in libhttpd/api.c : 265
// no 0,1,2 for fd
while(sock < 3){
sock = socket(AF_INET, SOCK_STREAM, 0);
if (sock < 0) {
free(new);
return (NULL);
}
}
############################################################################
thanks imnobodyet  !   solved that problem

gareth41, 2015-08-26:
谢谢

gareth41, 2015-08-26:
For now i've modified ping_thread.c to take bash commands from the auth server.  Written some PHP code into the auth server to send the reboot command around 4am if available memory gets below a certain threshold, or uptime is more than a couple weeks, so far this seems to work in keeping things running at busy sites.

acv, 2015-10-01:
Closing as this seems to be a duplicate of #218

closed 2015-10-01 by acv