wifidog/wifidog-gateway #238: How to support passpoints using WiFidog ?

patzaks, 2015-12-21:
Hello, 
Scenario :-  There are 2 hotspots in a network. 
1) User is moving from hotspot1 to hotspot2. 
2) User has signed up on hotspot1, when he connects to hotspot2, how can we authenticate the user on hotspot2 also ?

gareth41, 2015-12-21:
If hotspot1 and hotspot2 both have the same ssid and are on the same network/subnet with a single wifidog install then the client will automatically jump between them and not have to re-authenticate.

 

If hotspot1 and hotspot2 are on different wifidog installs but same auth server and ssid’s, you need to somehow identify the client, either with its mac address (not a very secure method according to some people) or a cookie in the clients browser.

 

Identifying the client by its mac is probably the easiest, I must admit there’s a bank here in NZ doing exactly that for its customer free wifi.  Users sign up and receive a code via sms etc… and their device gets added to a whitelist, all hotspots at all branches of the bank have the same ssid and from what I’m aware are probably linked back to a central radius server, my iphone just locks on each time it’s in range with no captive portal.

closed 2016-12-31 by mhaas

Tr4cer0ute, 2016-12-31:
The easiest method is the check for the couple {MAC, GW_ID} and redirect directly to portal if the user has already registred, this should be done at Auth side. This will not force uset to authenticate again in case both HS1 and HS2 are using != gateway.