wifidog/wifidog-gateway #243: Build wifidog for openwrt chaos_calmer/15.05

deivisonrpg, 2016-01-29:
Hi anyone build ipk package WiFiDog 1.3.0 for chaos-calmer/15.05 i try to build and don't make work.
The question WiFiDog 1.3.0 work on CC?
Tks all

deivisonrpg, 2016-01-29:
I have compile for CC using the toolchain has been generated and the ipk, however to install and try to run me back Ash: wifidog: not found.
[wifidog_1.3.0-1_ar71xx.ipk.zip](https://github.com/wifidog/wifidog-gateway/files/109860/wifidog_1.3.0-1_ar71xx.ipk.zip)

Can anyone confirm if this wifidog version works on openwrt CC?

deivisonrpg, 2016-01-29:
I try to use ipk from snapshots repository openwrt https://downloads.openwrt.org/snapshots/trunk/ar71xx/generic/packages/packages/wifidog_1.3.0-1_ar71xx.ipk and get same error. 
incompatibility for openwrt or wifidog?

mhaas, 2016-01-30:
That's... odd. I looked at the Openwrt IPK file and everything is in place. What happens if you execute /usr/bin/wifidog directly? Can you run ldd on the file?

mhaas, 2016-01-30:
Can you run:

    file /usr/bin/wifidog

It should report the interpreter, something like /lib/ld-uClibc.so.0 - does that file exist?

deivisonrpg, 2016-01-30:
I @mhaas 
If i try to run commands: file, ldd or execute directly from /usr/bin wifidog or wdctl i have same error:
-ash: XXXX: not found (XXXX is the file)

[image]

I try my ipk and openwrt ipk same thing.
If i install 1.2.1 ipk works fine.

mhaas, 2016-02-04:
That's.. weird. I don't have time to reproduce right now, but I will take a look next week (-ish).

demslob, 2016-02-13:
is there anyone with 15.05 ramips device who can give me ipk?

mgiller, 2016-09-27:
Hi

I have the same problem, have you find a solution?

tks

deivisonrpg, 2016-09-27:
Hi, no i give up.

turanmete-bhr, 2016-10-12:
Any update how you solved it ?

turanmete-bhr, 2016-10-12:
I also got the same situation after trying to compile 1.3.0.
Then I tried to compile 1.2.1 and use it. But I got built error like

protocol.c:534:10: warning: implicit declaration of function 'open' [-Wimplicit-function-declaration]
     fd = open(path, O_RDONLY);
          ^
protocol.c:534:21: error: 'O_RDONLY' undeclared (first use in this function)
     fd = open(path, O_RDONLY);
                     ^
protocol.c:534:21: note: each undeclared identifier is reported only once for each function it appears in
make[5]: *** [protocol.lo] Error 1
make[4]: *** [all-recursive] Error 1
make[3]: *** [all] Error 2

turanmete-bhr, 2016-10-17:
I think the problem is mainly about C compiler. When you are changing compiler to the older one (Menconfig - Advanced options - c library) still getting error but somehow its working I mean sending requests to auth server.

kuzjnvn, 2017-02-04:
anybody can make wifidog ipk for 14.07 openwrt ar71xx ?
thanks so much

mm1860, 2017-04-26:
./wifidog  ,command like this

closed 2019-05-16 by deivisonrpg