wifidog/wifidog-gateway #245: Wifidog Process Hangs

staurasi, 2016-02-09:
Hello, I have random wifidog hangs (about one every 2 or 3 days).

Wifidog version is 1.2.1
OpenWRT Chaos Calmer on an Ubiquiti AirRouter or Edgerouter LITE with more then 100 clients.
Auth server on a remote VPS.

This happens while wifidog checks for clients and it update the counters.
A child process is created but it never exits.
I ran wifidog in debug mode and with strace. The issue seem the same explained on this old thread :
http://listes.ilesansfil.org/pipermail/wifidog/2008-May/004512.html
But I found no valid answer there, and I have no clues if  the problem is related to the communication with the auth server or if something goes wrong while updating counters via iptables.

I'm I the only one facing this problem? Are >100 users too many?
Many thanks and let me know if you need anymore infos.

gareth41, 2016-02-09:
Can you try with a dedicated linux box (two network adapters) running wifidog and see if this problem still occurs?  100 clients is probably too much for small routers like the AirRouter.  I’ve run wifidog on the EnGenius EPG5000 with around 50 clients and not noticed any problems, although the EPG5000 does have significantly more ram (256mb) than the AirRouter and a much faster processor.  If you try the EPG5000 use openwrt trunk as the CC branch is only able to detect half the EPG5000 ram 128mb for some reason.

 

http://listes.ilesansfil.org/pipermail/wifidog/2008-May/004512.html

staurasi, 2016-02-10:
I cannot try with a dedicated linux box but I have already replaced the AirRouter with an Edgerouter LITE (2x500MHz cpu, 512MB ram, quite powerful). I've never seen high cpu or ram usage.

staurasi, 2016-02-12:
Any other clue? Please help.

mhaas, 2016-02-12:
Can you provide debug logs?

acv, 2016-02-12:
Or at least a full "ps aux" output showing the hung child process?

staurasi, 2016-02-15:
Yes, I'll provide debug logs, ps and strace outputs. It will take a few days.

Thanks for now and see you soon.

oleg-umnik, 2016-03-04:
I was able to reproduce this hang in openwrt environment forcing wifidog to repeatedly redirect unauthenticated user to portal and calling /wifidog/auth?token=XXX in parallel. Continuing investigation...

oleg-umnik, 2016-03-04:
#248

staurasi, 2016-03-09:
I tryed again during the last days, but in splash only mode to minimize the impact on the users (and to avoid a lot of login page requests from unauthenticated users smartphones.

Yesterday we reached 160 users with no issues. Instead this morning I had an hang just a few seconds after wifidog start (see logs, strace output and lsof output). I rebooted the router and and this second time everything is ok (118 users at the moment i write). I usually check with "wdctl status" a few seconds after launching, so maybe that triggered the issue? The logs say that auth server is not available, but that's seems odd since I can ping it and we have a few 
[lsof_child_process.txt](https://github.com/wifidog/wifidog-gateway/files/165091/lsof_child_process.txt)
[lsof_parent_process.txt](https://github.com/wifidog/wifidog-gateway/files/165088/lsof_parent_process.txt)

[ps_grep_wifidog.txt](https://github.com/wifidog/wifidog-gateway/files/165090/ps_grep_wifidog.txt)
[strace_child_process.txt](https://github.com/wifidog/wifidog-gateway/files/165089/strace_child_process.txt)

other routers connected to it.

oleg-umnik, 2016-03-09:
Your hang looks exactly like mine. Can you try patch from #248 pull request?

staurasi, 2016-03-16:
I created the openwrt build environment and replaced the wifidog src directory with the src zip downloaded from your branch oleg-umnik:hangfix, then compiled.
Is this the right way? I know this is a github/openwrt noob question.

Now the patched wifidog is running on the Edgerouter LITE (38 clients at the moment). I'll let you know.

andreasanta, 2016-04-07:
@staurasi did the fix solve the issue? I am also experiencing hangs with only 10-20 users, they really look like deadlocks.

For some reason the client list stays blocked and there is no way to unlock it. Maybe some thread dying with the lock state set? Any attempt to lock the client list results in a locked thread, so wdctl status locks, /wifidog/auth locks, and so on...

@oleg-umnik Do you think your patch can prevent that also?

staurasi, 2016-04-08:
I'm testing it in two location. Both with an EdgeRouter LITE. The first location has been running for more then one week and is running smooth with an average of 80 clients with normal login (no spalsh only). The second location (the one I care more) is in splash only mode with an avarage of 140 clients.
I run this second one only during daylight time because I want to be ready to reboot in case of problems. Today is the second day of testing and I didn't experience any hangs. Maybe it's still a little bit early to say, but I think it's working good!

andreasanta, 2016-04-08:
@saturasi I've managed to reproduce the issue on a Xiaomi Nano, by means of hitting the same "/wifidog/auth?token=XXX" URL many consecutive times, while unauthenticated users are being redirected to the portal, just like @oleg-umnik says. So, IMHO, it's a matter of time before it will happen again. Please monitor when you have more logins/sec **closely**, because that is when it is more likely to happen.

It's a clear race condition, since the client list mutex at some point gets locked forever, although I'm also evaluating some buffer overrun hitting the mutex struct.

staurasi, 2016-04-08:
@andreasanta That would explain why sometimes I had an hang just a few seconds after wifidog start. In splash only mode, starting wifidog with clients already connected to the wifi network results in an high login/sec ratio. I also hope that @oleg-umnik patch can prevent this. In the meantime I'll keep monitoring very closely.

andreasanta, 2016-04-10:
After thorough investigation, I've managed to reproduce the hang consistently:

1. Start WifiDog
2. Plug in a laptop on the LAN port, and authenticate
3. Copy the auth token for your laptop shown from "wdctl status" and rebuild the auth url, for instance http://192.168.1.1/widifog/auth?token=token
4. Use apache bench (ab) on your laptop to stress test wifidog against the auth URL with a concurrency of at least 10, in my case I would use "ab -s 120 -n 50  -c 10"
5. While ab is running, connect to the wifi with another device, try to navigate here and there until you get caught by the portal.

If you repeat point 4 and 5 a few times, you will see the log hangs at:

> wifidog[14647]: Waiting for PID 15138 to exit 

More precisely, as soon as the child process is executed and _cleanup_fds_ is called, it tries and free some memory (an fd_list_t *) and the process hangs waiting on a mutex. Since the process has the client list mutex locked when hanging, no other process is able lock the client list. So you get a total wifidog hang.

So @oleg-umnik patch fixed the issue for me. Please merge PR #248 to fix the issue.