wifidog/wifidog-gateway #250: cant get gateway ip - socket(): Invalid argument

gareth41, 2016-04-08:
Running openwrt trunk from 21st March and getting the following:

socket(): Invalid argument
Could not get IP address information of br-lan, exiting...

Looks like the problem is util.c
if ((sockd = socket(AF_INET, SOCK_RAW, htons(0x8086))) < 0) {

KazukiShimada, 2016-04-09:
waiting for merging.
https://github.com/wifidog/wifidog-gateway/pull/244

kuzjnvn, 2017-02-09:
i have problem with thiss... wifidog 1.2.1-1
OpenWrt Chaos Calmer 15.05.1 r49474 - ar71xx
anybody can help

nuwanarti, 2018-06-27:
I have this same issue. i installed WiFiDog version 1.3.0 from opkg , and my lede version is Linux LEDE 4.4.92 #0

nguyennhatdung, 2019-07-03:
i got the same issue 

i had try br-lan,eth0,eth0.1,eth0.2,... but none of it working

any solution ?